espressif/esp-csi #1: Unable to enter commands into the Interactive Shell

zhouzhi1, 2021-02-28:
Steps to test CSI                                |
 |                                                                     |
 |  1. Print 'help' to gain overview of commands                       |
 |     csi> help                                                       |
 |  2. Start SoftAP or Sta on another ESP32/ESP32S2/ESP32C3            |
 |     csi> sta <ssid> <pwd>                                           |
 |     csi> ap <ssid> <pwd>                                            |
 |  3. Run ping to test                                                |
 |     csi> ping <ip> -c <count> -i <interval>                         |
 |  4. Configure CSI parameters                                        |
 |     csi> csi -m <mac(xx:xx:xx:xx:xx:xx)> -l <len>                   |
 |  5. Configure CSI parameters                                        |
 |     csi> detect -a <absolute_threshold> -r <relative_threshold>

This message keeps repeating itself.

EspHuifeng, 2021-03-03:
You might want to qualify not working, describe what you were doing, and if any, post the serial terminal that you are using.

zhouzhi1, 2021-03-03:
I have already got the answer from another issue. Thanks for your reply.

zhouzhi1, 2021-03-03:
I (447698) app_main: <14> **time: 23110 ms**, rssi: -30, corr: 0.288, std: 0.116, std_avg: 0.000, std_max: 0.000, threshold: 0.017/1.500, trigger: 1/0, free_heap: 135340/168904
Sorry, I'm new hand. Although the project ran successfully, it took a long time. How to reduce time value.

JunyiXb, 2021-07-26:
How to get the csi ternminal through  ESP-IDF PowerShell ,I Use the command  "idf.py -p com9 monitor" to get  csi terminal but I can not  write command  "ap csi_soft" to change the mode of ESP32,and disappare from screen?

closed 2023-05-31 by zhanzhaocheng