espressif/esp-csi #2: ESP32运行会看门狗复位

AzollaGit, 2021-03-02:
运行环境：IDF master分支。 ESP32-WROOM-32模组。
运行DEBUG:
[image]

EspHuifeng, 2021-03-03:
Please use the specific IDF commit described here. https://github.com/espressif/esp-csi/blob/79cdc17f65cedebe26f26160b76e13c53c150a78/CMakeLists.txt#L1

Lign17314, 2021-03-09:
我也遇见了这个问题，用的也是master分支，请问应该如何解决，
starting ap is ok until others connect.but if one connect，the modul e will reset. 
when sta connect to wifi, it will reset if the password is right.

Lign17314, 2021-03-09:
我已经可以正常运行了，参照的.gitlab-ci.yml文件中几个命令将esp-idf更新到了特定的版本，使用的命令如下
到esp-idf文件夹中
  - cd esp-idf
  - git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06
  - git submodule update --init --recursive
  - source tools/ci/setup_python.sh
  - source tools/ci/configure_ci_environment.sh
到esp-csi文件夹中
    - cd examples/console_test
    - idf.py set-target esp32
    - idf.py build
  然后下载就可以了，可以正常连接

我不是怎么会用git的相关命令，但我猜测应该是特定esp-idf版本的原因，现在能稳定使用的只有那一个版本

Lign17314, 2021-03-09:
> 运行环境：IDF master分支。 ESP32-WROOM-32模组。
> 运行DEBUG:
> [image]

我已经可以正常运行了，参照的.gitlab-ci.yml文件中几个命令将esp-idf更新到了特定的版本，使用的命令如下
到esp-idf文件夹中

cd esp-idf
git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06
git submodule update --init --recursive
source tools/ci/setup_python.sh
source tools/ci/configure_ci_environment.sh
到esp-csi文件夹中
cd examples/console_test
idf.py set-target esp32
idf.py build
然后下载就可以了，可以正常连接
我不是怎么会用git的相关命令，但我猜测应该是特定esp-idf版本的原因，现在能稳定使用的只有那一个版本

R88624332, 2021-06-15:
> 我已经可以正常运行了，参照的.gitlab-ci.yml文件中几个命令将esp-idf更新到了特定的版本，使用的命令如下
> 到esp-idf文件夹中
>
> cd esp-idf
> git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06
> git submodule update --init --recursive
> source [...]

兄弟我也是用ESP32遇到同样的问题，但是我照你的方法用`git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06`代码要报错，不知道你有遇到这种情况吗
```
PS E:\esp_test\esp-idf> git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06
error: Your local changes to the following files would be overwritten by checkout:
        docs/build_docs.py
        docs/test/test_sphinx_idf_extensions.py
        install.fish
        install.sh
        tools/idf_monitor.py
        tools/idf_size.py
        tools/idf_tools.py
        tools/ldgen/ldgen.py
        tools/ldgen/test/test_entity.py
        tools/ldgen/test/test_fragments.py
        tools/ldgen/test/test_generation.py
        tools/ldgen/test/test_output_commands.py
Please commit your changes or stash them before you switch branches.
Aborting
```

closed 2023-05-31 by zhanzhaocheng