espressif/esp-csi #3: How to get CSI data in monitor mode?

kingaza, 2021-03-09:
It is such a good example. Now I can get CSI data by using `ping `command.
However, I notice that the time interval is not fixed, from 0.01 to 0.1 second. I have to do interpolation or even resampling before signal processing.
Therefore, I would like to try command mode to get a formated time serires data.

thanks for your help!

zhanzhaocheng, 2021-03-10:
The information printed by the serial port contains [the time](https://github.com/espressif/esp-csi/blob/master/examples/console_test/main/app_main.c#L61-L79), you can format the time with the ‘strftime’ function

kingaza, 2021-03-17:
As in the example, the time gap of the data is not fixed.

```
time_diff = np.diff(df.local_timestamp)

time_diff.min(), time_diff.max()
Out[3]: (2623, 111798)

time_diff.std(), time_diff.mean()
Out[4]: (7791.2255138132, 14511.423648247177) 
```
Is there a method to set an accurate time gap?

kingaza, 2021-03-19:
close this issue, since I can do interpolation anyway...

closed 2021-03-19 by kingaza