espressif/esp-csi #4: How to deal with first_word_invalid?

roger-, 2021-03-13:
Does it only apply to the LLTF data or will HT-LTF, etc. be affected?

zhanzhaocheng, 2021-03-15:
Receiving sub-carrier: LLTF, HT-LTF, STBC-HT-LTF,
The data content of the sub-carrier depends on the sending device
[image]

roger-, 2021-03-15:
So if LLTF, HT-LTF and  STBC-HT-LTF are all received then only the first two frequency bins of LLTF will be invalid?

zhanzhaocheng, 2021-03-15:
If first_word_invalid field of wifi_csi_info_t is true, it means that the first four bytes of CSI data is invalid due to a hardware limitation in ESP32.

roger-, 2021-03-15:
That means the first four bytes of `wifi_csi_info_t.buf` right (which is LLTF if you are getting all three)? 

As opposed to the first four bytes of each of LLTF, HT-LTF and STBC-HT-LTF within that variable.

zhanzhaocheng, 2021-03-15:
It only means that the first four bytes of `wifi_csi_info_t.buf` are invalid

closed 2021-03-15 by roger-

NiyamPoudel, 2021-07-24:
Hello. I have been working on Human Fall Detection using CSI Signals. However, I dont have prior knowledge on the subject. I have been trying to understand the documentation but I have a lot of questions. I followed https://github.com/StevenMHernandez/ESP32-CSI-Tool and got a matrix of CSI values. 

```[112 -128 6 0 0 0 0 0 0 0 0 0 3 9 2 10 1 12 0 14 1 14 1 16 2 16 3 16 3 16 4 15 5 15 5 14 5 13 4 12 2 11 1 11 0 11 -1 11 -2 12 -3 13 -3 14 -3 15 -2 16 -1 17 0 17 1 17 0 0 2 15 2 14 2 14 1 12 0 12 -1 12 -2 12 -4 12 -4 13 -4 15 -4 16 -4 17 -3 18 -3 19 -1 19 0 19 1 18 2 17 1 16 2 15 0 14 -1 14 -2 13 -3 14 -4 15 -6 16 0 0 0 0 0 0 0 0 0 0 ]````

Could anyone help me understand how I would read this data or point me to a useful resource. Thanks a lot.

hz1767110017, 2021-07-24:
What i understand is every two number of this csi data represents the real and the imaginary parts of the frequency response of each subcarrier in IQ channel, you can get the amplitude and phase from it.