espressif/esp-csi #6: esp - csi code compile link error when using vs code.

zkpingguo, 2021-03-16:
........
[927/928] Linking CXX executable console_test.elf
FAILED: console_test.elf
cmd.exe /C "cd . && C:\Users\zhaokai\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\console_test.elf.rsp  -o console_test.elf  && cd ."
c:/users/zhaokai/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/ESP/esp-csi-master/esp-csi-master/lib/esp32/libesp-csi.a(esp_radar.c.obj):(.literal.csi_data_combined_task+0x2c): undefined reference to `xQueueReceive'
c:/users/zhaokai/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/ESP/esp-csi-master/esp-csi-master/lib/esp32/libesp-csi.a(esp_radar.c.obj): in function `csi_data_combined_task':
/home/zhanghuifeng/esp/esp-csi/src/esp_radar.c:439: undefined reference to `xQueueReceive'
c:/users/zhaokai/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/ESP/esp-csi-master/esp-csi-master/lib/esp32/libesp-csi.a(esp_radar.c.obj): in function `csi_data_handle_task':
/home/zhanghuifeng/esp/esp-csi/src/esp_radar.c:348: undefined reference to `xQueueReceive'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
终端进程“C:\windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build .”已终止，退出代码: 1。

zhanzhaocheng, 2021-03-17:
The currently dependent esp-idf version is 6e776946d01ec0d081d09000c36d23ec1d318c06, please switch to this version
```shell
cd esp-idf
git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06
git submodule update --init --recursive
```

zkpingguo, 2021-03-30:
thanks , I have solved this issue.

zhy102, 2022-03-10:
> ```shell
> git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06
> ```
Hello, i have the same issue. However,  i called  "cd esp-idf " and "git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06```, the errors:
``` fatal: reference is not a tree: 6e776946d01ec0d081d09000c36d23ec1d318c06```

The errors as follows when i ran "idf.py flash":

C:\Espressif\frameworks\esp-idf-v4.3.2\esp-csi-master\examples\console_test>idf.py flash
```
Executing action: flash
Serial port COM13
Connecting....
Detecting chip type... ESP32
Running ninja in directory c:\espressif\frameworks\esp-idf-v4.3.2\esp-csi-master\examples\console_test\build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32
??? Linux ? Windows ?????????????
?????? Microsoft Store ??????:
https://aka.ms/wslstore
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v4.3.2/esp-csi-master/examples/console_test/sdkconfig
C:\Users\ADMINI~1\AppData\Local\Temp\confgen_tmp2_fq_wu7:5: warning: 'N' is not a valid value for the bool symbol ESP32_WIFI_AMPDU_TX_ENABLED (defined at C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_wifi/Kconfig:128). Assignment ignored.
C:\Users\ADMINI~1\AppData\Local\Temp\confgen_tmp2_fq_wu7:6: warning: 'N' is not a valid value for the bool symbol ESP32_WIFI_AMPDU_RX_ENABLED (defined at C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_wifi/Kconfig:146). Assignment ignored.
Loading defaults file C:/Espressif/frameworks/esp-idf-v4.3.2/esp-csi-master/examples/console_test/sdkconfig.defaults...
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- App "console_test" version: v4.3.2
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/esp-csi-master/examples/console_test/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp32/ld/esp32.peripherals.ld
??? Linux ? Windows ?????????????
?????? Microsoft Store ??????:
https://aka.ms/wslstore
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmd_nvs cmd_system cmock coap console cxx driver efuse esp-csi-master esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v4.3.2/components/app_trace C:/Espressif/frameworks/esp-idf-v4.3.2/components/app_update C:/Espressif/frameworks/esp-idf-v4.3.2/components/asio C:/Espressif/frameworks/esp-idf-v4.3.2/components/bootloader C:/Espressif/frameworks/esp-idf-v4.3.2/components/bootloader_support C:/Espressif/frameworks/esp-idf-v4.3.2/components/bt C:/Espressif/frameworks/esp-idf-v4.3.2/components/cbor C:/Espressif/frameworks/esp-idf-v4.3.2/examples/system/console/components/cmd_nvs C:/Espressif/frameworks/esp-idf-v4.3.2/examples/system/console/components/cmd_system C:/Espressif/frameworks/esp-idf-v4.3.2/components/cmock C:/Espressif/frameworks/esp-idf-v4.3.2/components/coap C:/Espressif/frameworks/esp-idf-v4.3.2/components/console C:/Espressif/frameworks/esp-idf-v4.3.2/components/cxx C:/Espressif/frameworks/esp-idf-v4.3.2/components/driver C:/Espressif/frameworks/esp-idf-v4.3.2/components/efuse C:/Espressif/frameworks/esp-idf-v4.3.2/esp-csi-master C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp-tls C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp32 C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_adc_cal C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_common C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_eth C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_event C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_gdbstub C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_hid C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_http_client C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_http_server C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_https_ota C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_https_server C:/Espressif/frameworks/esp-idf-v4.3.2/components/esp_hw_support C:/Espressif/frameworks/esp-idf- [... cut ...] 19xchacha20poly1305.c.obj   -c C:/Espressif/frameworks/esp-idf-v4.3.2/components/libsodium/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c
ccache: error: Failed to create temporary file for esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_box/curve25519xchacha20poly1305/box_seal_curve25519xchacha20poly1305.c.obj: No such file or directory
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

zhy102, 2022-03-10:
Hello , I also have this issue. 
 However, i called "cd esp-idf " and "git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06, the errors:
```fatal: reference is not a tree: 6e776946d01ec0d081d09000c36d23ec1d318c06.```
What should I do？

closed 2023-05-31 by zhanzhaocheng