espressif/esp-csi #7: CSI data format

kingaza, 2021-03-17:
Hello,

I want to make sure the CSI data format.

For example, the csi data:
`CSI_DATA,0,ac:67:b2:53:8f:31,-26,11,1,7,1,1,1,0,0,0,0,-88,0,13,2,51236189,0,66,0,384,1,"[66,32,4,0,0,0,0,0,0,0,3,0,15,0,15,-1,15,-1,15,-1,15,-1,15,-1,15,-1,15,-2,15,-2,14,-2,14,-2,14,-3,15,-3,15,-3,15,-3,15,-4,15,-4,15,-4,15,-4,15,-4,15,-4,15,-5,14,-5,14,-6,15,-6,15,-7,7,-4,15,-7,15,-7,15,-7,15,-7,16,-7,16,-7,16,-7,17,-7,17,-8,17,-8,17,-8,18,-8,18,-8,18,-8,19,-8,19,-8,19,-7,20,-7,20,-6,20,-6,20,-5,21,-4,20,-4,21,-4,21,-3,21,-2,5,-1,0,0,0,0,0,0,0,0,1,0,4,0,28,3,28,2,28,2,28,2,28,0,28,0,28,0,28,-1,28,-1,28,-1,28,-2,28,-2,28,-3,28,-3,28,-3,28,-4,28,-4,28,-5,28,-5,28,-6,28,-7,28,-7,28,-7,28,-8,28,-8,28,-9,28,-10,28,-10,28,-11,29,-11,29,-11,29,-12,29,-13,29,-13,30,-13,30,-14,30,-14,31,-14,31,-15,32,-15,32,-15,33,-15,34,-15,34,-15,35,-15,35,-15,36,-14,37,-13,37,-13,38,-12,38,-11,39,-10,39,-9,39,-8,40,-7,40,-5,40,-4,12,-2,2,-1,-1,0,-2,0,-1,-1,0,0,0,0,-1,-1,-1,-1,-1,-1,-4,5,-12,23,-13,23,-14,22,-14,22,-15,22,-15,22,-16,21,-16,21,-16,20,-17,20,-17,20,-17,20,-17,19,-18,19,-18,19,-17,19,-17,19,-17,18,-17,18,-17,19,-17,18,-16,18,-16,18,-16,18,-15,19,-15,19,-16,18,-15,19,-15,19,-14,19,-13,20,-13,19,-12,19,-12,20,-12,21,-12,21,-11,21,-11,21,-10,22,-10,22,-9,22,-9,23,-9,23,-8,23,-8,23,-8,24,-7,24,-7,24,-7,25,-6,25,-6,25,-5,26,-5,26,-6,27,-6,27,-5,27,-4,27,-1,3]"
`
sig_mode=1
secondary_channel= 2
bandwidth = 2
stbc = 1
len = 384
first_word =1

So, this data shall be 
[image]
Then
LLTF: -64 ~ -1
HT-LLF: 0 ~ 63, -64 ~ -1
So, the first 64*2=128 bytes are LLTF, and the last (64+64)*2=256 bytes are HT-LLF?

And I have still a question that why so much channels are ignored in analisys_csi_data.py?
```
select_list = []
select_list += [i for i in range(5, 31)]
select_list += [i for i in range(33, 58)]
select_list += [i for i in range(66, 122)]
select_list += [i for i in range(123, 191)]
select_list.remove(128)
```

thx.

kingaza, 2021-03-17:
Another confusion is,
STBC or nonSTBC is it in this case?

zhanzhaocheng, 2021-03-17:
Only multi-antenna devices will send data to include STBC. ESP32 does not support multi-antenna packet transmission, but it can receive data from multi-antenna devices.
Since we use ESP32 devices by default for testing, the script only deals with `no STBC` information

kingaza, 2021-03-19:
@zhanzhaocheng , so the data is actually nonSTBC information.
So, according to 
```
LLTF: -64 ~ -1
HT-LLF: 0 ~ 63, -64 ~ -1
```
is this correct?
**the first 64*2=128 bytes are LLTF, and the last (64+64)*2=256 bytes are HT-LLF?**

thx again.

zhanzhaocheng, 2021-03-22:
The received byte is 384, which is distributed in this way, but the range of different channels and bandwidths is different. Please refer to::https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information

kingaza, 2021-03-23:
上面那个图就是从参考文档抓下来的。现在还不是特别清楚CSI的数据格式。

比如现在384个字节，那么就是192个复数数据。按照参考文档说的
LLTF有64个，HT-LLF有128个
那么是否这192个数据中，前面64个是LLTF，后面128个是HT-LLF？

另外在代码里面提到
```
select_list = []
select_list += [i for i in range(5, 31)]
select_list += [i for i in range(33, 58)]
select_list += [i for i in range(66, 122)]
select_list += [i for i in range(123, 191)]
select_list.remove(128)
```
为啥要这么选择呢？

closed 2023-05-31 by zhanzhaocheng