espressif/esp-csi #8: connect ap crash

zkpingguo, 2021-03-30:
csi> E (16997) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (16997) task_wdt:  - IDLE (CPU 0)
E (16997) task_wdt: Tasks currently running:
E (16997) task_wdt: CPU 0: wifi
E (16997) task_wdt: CPU 1: IDLE
E (16997) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400FF4AB:0x3FFB0810 0x40082871:0x3FFB0830 0x4010D5E1:0x3FFC0320 0x4011A7EF:0x3FFC0340 0x40090441:0x3FFC0460 0x400904F9:0x3FFC04C0 0x400948DD:0x3FFC04E0 0x400921E8:0x3FFC0500 0x4008CCDD:0x3FFC0530 

E (16997) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008496D:0x3FFB0E10 0x40082871:0x3FFB0E30 0x4000BFED:0x3FFBBD50 0x4008CFBE:0x3FFBBD60 0x400FF74F:0x3FFBBD80 0x400FF75B:0x3FFBBDB0 0x400D8A11:0x3FFBBDD0 0x4008A995:0x3FFBBDF0 0x4008CCDD:0x3FFBBE10

Jiangyafeng, 2021-04-01:
@zkpingguo To set the idf environment variable, you need to commit on the IDF master branch: 6e776946d01ec0d081d09000c36d23ec1d318c06. You can also see the version number in CMakeLists.txt.

Jiangyafeng, 2021-04-01:
@zkpingguo Please also update the warehouse.

closed 2023-05-31 by zhanzhaocheng