espressif/esp-csi #9: how to build the project

killer-p, 2021-04-01:
my idf version is v4.4-beta
I run `idf.py` build,and then it print
---------------------------------------------------------------------
root@dragon:/esp/esp-adf/examples/esp-csi# idf.py build
WARNING: Support for Python 2 is deprecated and will be removed in future versions.
Executing action: all (aliases: build)
Running cmake in directory /esp/esp-adf/examples/esp-csi/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /esp/esp-adf/examples/esp-csi"...
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

fatal: not a git repository (or any of the parent directories): .git
Currect esp-idf version is
check_idf_version.sh: 8: [[: not found
CMake Error at CMakeLists.txt:4 (idf_build_get_property):
  Unknown CMake command "idf_build_get_property".

CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.16)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring incomplete, errors occurred!
See also "/esp/esp-adf/examples/esp-csi/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

---------------------------------------------------------

then I run `cmake .` it print:
------------------------------------------------------------
root@dragon:/esp/esp-idf# cd /esp/esp-adf/examples/esp-csi
root@dragon:/esp/esp-adf/examples/esp-csi# cmake .
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

fatal: not a git repository (or any of the parent directories): .git
Currect esp-idf version is
check_idf_version.sh: 8: [[: not found
CMake Error at CMakeLists.txt:4 (idf_build_get_property):
  Unknown CMake command "idf_build_get_property".

-----------------------------------------------------------------------

how can I build the example?

Jiangyafeng, 2021-04-01:
@killer-p  
From your log, it is because the idf environment is not installed properly.

Need to pay attention to the following points:
1. Update python version to python3.

2. To set the idf environment variable, you need to commit on the master branch: 6e776946d01ec0d081d09000c36d23ec1d318c06. You can also see the version number in CMakeLists.txt.

3. Set up a connection for the idf environment: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#installation-step-by-step

killer-p, 2021-04-01:
> @killer-p
> From your log, it is because the idf environment is not installed properly.
>
> Need to pay attention to the following points:
>
> 1. Update python version to python3.
> 2. To set the idf [...]

thank you!I checkout idf to 6e776946d01ec0d081d09000c36d23ec1d318c06 and update idf environment.But I still meet some problem when building.Here is the log:
------------------------------------------------------------------
root@dragon:/esp/esp-adf/examples/esp-csi/examples/console_test# idf.py build
Executing action: all (aliases: build)
Running cmake in directory /esp/esp-adf/examples/esp-csi/examples/console_test/build
Executing "cmake -G 'Unix Makefiles' -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /esp/esp-adf/examples/esp-csi/examples/console_test"...
CMake Error at CMakeFiles/3.16.3/CMakeSystem.cmake:6 (include):
  include could not find load file:

    /esp/idf/esp-idf/tools/cmake/toolchain-esp32.cmake
Call Stack (most recent call first):
  /esp/idfv4.4/esp-idf/tools/cmake/project.cmake:308 (__project)
  CMakeLists.txt:23 (project)

-- Building ESP-IDF components for target esp32
Currect esp-idf version is 6e776946d01ec0d081d09000c36d23ec1d318c06
-- Project sdkconfig file /esp/esp-adf/examples/esp-csi/examples/console_test/sdkconfig
Loading defaults file /esp/esp-adf/examples/esp-csi/examples/console_test/sdkconfig.defaults...
/tmp/confgen_tmpf9vwhnmk:5: warning: 'N' is not a valid value for the bool symbol ESP32_WIFI_AMPDU_TX_ENABLED (defined at /esp/idfv4.4/esp-idf/components/esp_wifi/Kconfig:128). Assignment ignored.
/tmp/confgen_tmpf9vwhnmk:6: warning: 'N' is not a valid value for the bool symbol ESP32_WIFI_AMPDU_RX_ENABLED (defined at /esp/idfv4.4/esp-idf/components/esp_wifi/Kconfig:146). Assignment ignored.
CMake Warning (dev) at /esp/idfv4.4/esp-idf/components/mbedtls/CMakeLists.txt:113 (target_sources):
  Policy CMP0076 is not set: target_sources() command converts relative paths
  to absolute.  Run "cmake --help-policy CMP0076" for policy details.  Use
  the cmake_policy command to set the policy and suppress this warning.

  A private source from a directory other than that of target "mbedcrypto"
  has a relative path.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- App "console_test" version: 5ef2890
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /esp/esp-adf/examples/esp-csi/examples/console_test/esp-idf/esp32/esp32_out.ld
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /esp/idfv4.4/esp-idf/components/esp32/ld/esp32.peripherals.ld
Currect esp-idf version is 6e776946d01ec0d081d09000c36d23ec1d318c06
check_idf_version.sh: 8: [[: not found
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmd_nvs cmd_system cmock coap console cxx driver efuse esp-csi esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_local_ctrl esp_netif esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /esp/idfv4.4/esp-idf/components/app_trace /esp/idfv4.4/esp-idf/components/app_update /esp/idfv4.4/esp-idf/components/asio /esp/idfv4.4/esp-idf/components/bootloader /esp/idfv4.4/esp-idf/components/bootloader_support /esp/idfv4.4/esp-idf/components/bt /esp/idfv4.4/esp-idf/components/cbor /esp/idfv4.4/esp-idf/examples/system/console/components/cmd_nvs /esp/idfv4.4/esp-idf/examples/system/console/components/cmd_system /esp/idfv4.4/esp-idf/components/cmock /esp/idfv4.4/esp-idf/components/coap /esp/idfv4.4/esp-idf/components/console /esp/idfv4.4/esp-idf/components/cxx /esp/idfv4.4/esp-idf/components/driver /esp/idfv4.4/esp-idf/components/efuse /esp/esp-adf/examples/esp-csi /esp/idfv4.4/esp-idf/components/esp-tls /esp/idfv4.4/esp-idf/components/esp32 /esp/idfv4.4/esp-idf/components/esp_adc_cal /esp/idfv4.4/esp-idf/components/esp_common /esp/idfv4.4/esp-idf/components/esp_eth /esp/idfv4.4/esp-idf/components/esp_event /esp/idfv4.4/esp-idf/components/esp_gdbstub /esp/idfv4.4/esp-idf/components/esp_hid /esp/idfv4.4/esp-idf/components/esp_http_client /esp/idfv4.4/esp-idf/components/esp_http_server /esp/idfv4.4/esp-idf/components/esp_https_ota /esp/idfv4.4/esp-idf/components/esp_https_server /esp/idfv4.4/esp-idf/components/esp_hw_support /esp/idfv4.4/esp-idf/components/esp_ipc /esp/idfv4.4/esp-idf/components/esp_local_ctrl /esp/idfv4.4/e [... cut ...] v4.4/esp-idf/components/wifi_provisioning /esp/idfv4.4/esp-idf/components/wpa_supplicant /esp/idfv4.4/esp-idf/components/xtensa
Project commit: 5ef2890
-- Configuring incomplete, errors occurred!
See also "/esp/esp-adf/examples/esp-csi/examples/console_test/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
root@dragon:/esp/esp-adf/examples/esp-csi/examples/console_test# 
------------------------------------------------------------------------------------

what is the problem about this?

Jiangyafeng, 2021-04-01:
@killer-p 
Do you have a git submoudle update --init --recurse operation? 
In order to ensure that the installation package is available. Please proceed as follows:

Command:
cd esp-idf/
export IDF_PATH=~/path/to/esp-idf/
./add_path.sh
./install.sh
./export.sh
cd esp-csi/examples/console_test

Then go to compile.

NiyamPoudel, 2021-08-03:
I am getting the following error when I enter `cmake .` Could you please help @Jiangyafeng .
```
nimai@nimai:~/esp-csi/examples/console_test$ export PATH="$PATH:$HOME/esp/xtensa-esp32-elf/bin"
nimai@nimai:~/esp-csi/examples/console_test$ export IDF_PATH=~/esp/esp-idf
nimai@nimai:~/esp-csi/examples/console_test$ cmake .
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Checking Python dependencies...
Python requirements from /home/nimai/esp/esp-idf/requirements.txt are satisfied.
-- IDF_TARGET not set, using default target: esp32
-- Building for target esp32
-- ccache will be used for faster builds
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/nimai/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/nimai/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/nimai/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/nimai/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/nimai/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Currect esp-idf version is 143d26aa49df524e10fb8e41a71d12e731b9b71d
** Warning **
The version of esp-idf is not expected, please set it to 6e776946d01ec0d081d09000c36d23ec1d318c06, otherwise it may not run correctly
CMake Error at /home/nimai/esp-csi/CMakeLists.txt:4 (idf_build_get_property):
  Unknown CMake command "idf_build_get_property".
Call Stack (most recent call first):
  /home/nimai/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:107 (include)
  /home/nimai/esp/esp-idf/tools/cmake/scripts/expand_requirements.cmake:217 (expand_component_requirements)

CMake Error at /home/nimai/esp/esp-idf/tools/cmake/idf_functions.cmake:338 (message):
  Failed to expand component requirements
Call Stack (most recent call first):
  /home/nimai/esp/esp-idf/tools/cmake/project.cmake:173 (idf_import_components)
  CMakeLists.txt:23 (project)

-- Configuring incomplete, errors occurred!
See also "/home/nimai/esp-csi/examples/console_test/CMakeFiles/CMakeOutput.log".
See also "/home/nimai/esp-csi/examples/console_test/CMakeFiles/CMakeError.log".
```

Jiangyafeng, 2021-08-03:
@NiyamPoudel  You need to switch the version to idf commit as:6e776946d01ec0d081d09000c36d23ec1d318c06， At the same time, it is recommended that you use idf's own library to help you install it. The process operation is explained above.

closed 2023-05-31 by zhanzhaocheng