espressif/esp-csi #12: console_test 编译不过（配合moonlight开发板）

cuityanxi, 2021-04-20:
esptool.py v3.0
Generated D:/ESP/esp-idf/examples/esp-csi-master/examples/console_test/build/bootloader/bootloader.bin
[939/940] Linking CXX executable console_test.elf
FAILED: console_test.elf
cmd.exe /C "cd . && D:\ESP\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\console_test.elf.rsp  -o console_test.elf  && cd ."
d:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/ESP/esp-idf/examples/esp-csi-master/lib/esp32/libesp-csi.a(esp_radar.c.obj):(.literal.csi_data_combined_task+0x2c): undefined reference to `xQueueReceive'
d:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/ESP/esp-idf/examples/esp-csi-master/lib/esp32/libesp-csi.a(esp_radar.c.obj): in function `csi_data_combined_task':
/home/zhanghuifeng/esp/esp-csi/src/esp_radar.c:439: undefined reference to `xQueueReceive'
d:/esp/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:/ESP/esp-idf/examples/esp-csi-master/lib/esp32/libesp-csi.a(esp_radar.c.obj): in function `csi_data_handle_task':
/home/zhanghuifeng/esp/esp-csi/src/esp_radar.c:348: undefined reference to `xQueueReceive'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

zztiger123, 2021-04-20:
您好, ESP-CSI 需要在特定的 idf 版本上运行， commit 为 6e776946d01ec0d081d09000c36d23ec1d318c06。你需要 `git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06` 来切换到指定的 idf 版本。

cuityanxi, 2021-04-20:
esp-idf-tools-setup-2.3 安装的master，如何切过去呢

Jiangyafeng, 2021-04-20:
@cuityanxi 建议您切换到linux下 去开发，你这个工具链太老了，是不支持的。

cuityanxi, 2021-05-07:
../main/qcloud_light.h:13:10: fatal error: esp_qcloud_log.h: No such file or directory
 #include "esp_qcloud_log.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/wifi/csi/esp-csi/examples/connect_qcloud$ 
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/wifi/csi/esp-csi/examples/connect_qcloud$ 
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/wifi/csi/esp-csi/examples/connect_qcloud$ cd ..
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/wifi/csi/esp-csi/examples$ cd ..
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/wifi/csi/esp-csi$ cd ..
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/wifi/csi$ cd ..
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/wifi$ cd ..
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples$ cd ..
jiangyanxi@GMPC2020050018:~/esp/esp-idf$ git show
commit 6e776946d01ec0d081d09000c36d23ec1d318c06 (HEAD, develop)
Merge: c4346abfb d34acd0d8
Author: Mahavir Jain <mahavir@espressif.com>
Date:   Mon Feb 1 13:22:30 2021 +0800

    Merge branch 'bugfix/update_howsmyssl_cert' into 'master'
    
    examples/protocols: update www.howsmyssl.com certificate to fix test failures
    
    See merge request espressif/esp-idf!12202

还是有问题呢？

cuityanxi, 2021-05-07:
以解决

closed 2021-05-07 by cuityanxi