espressif/esp-csi #13: git clone 以及menuconfig错误问题

cuityanxi, 2021-04-30:
问题：拉idf报错“步骤1”，     menuconfig报错 “步骤3”

每次拉idf 都会出现：
步骤1～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/nghttp/nghttp2'...
remote: Enumerating objects: 41013, done.        
remote: Counting objects: 100% (284/284), done.        
remote: Compressing objects: 100% (129/129), done.        
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: 过早的文件结束符（EOF）
fatal: index-pack 失败
fatal: 无法克隆 'https://github.com/nghttp2/nghttp2.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/nghttp/nghttp2'
克隆 'components/nghttp/nghttp2' 失败。按计划重试
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/protobuf-c/protobuf-c'...
remote: Enumerating objects: 3640, done.        
remote: Counting objects: 100% (107/107), done.        
remote: Compressing objects: 100% (72/72), done.        
remote: Total 3640 (delta 55), reused 57 (delta 30), pack-reused 3533        
接收对象中: 100% (3640/3640), 1.47 MiB | 652.00 KiB/s, 完成.
处理 delta 中: 100% (2322/2322), 完成.
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/spiffs/spiffs'...
remote: Enumerating objects: 1497, done.        

.....
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/esp_wifi/lib'...
fatal: unable to access 'https://github.com/espressif/esp32-wifi-lib.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
fatal: 无法克隆 'https://github.com/espressif/esp32-wifi-lib.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/esp_wifi/lib'
第二次尝试克隆 'components/esp_wifi/lib' 失败，退出

步骤 2～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～

git checkout 6e776946d01ec0d081d09000c36d23ec1d318c06 到csi的head后（问题恢复： connect_qcloud 编译不过（配合moonlight开发板） #12 ）

步骤3～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～～
     menuconfig报错
jiangyanxi@GMPC2020050018:~/esp/esp-idf/examples/bluetooth/esp_ble_mesh/aligenie_demo$ idf.py menuconfig
WARNING: Support for Python 2 is deprecated and will be removed in future versions.
Executing action: menuconfig
Running cmake in directory /home/jiangyanxi/esp/esp-idf/examples/bluetooth/esp_ble_mesh/aligenie_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/jiangyanxi/esp/esp-idf/examples/bluetooth/esp_ble_mesh/aligenie_demo"...
-- Found Git: /usr/bin/git (found version "2.17.1") 
CMake Warning at /home/jiangyanxi/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/asio/asio is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/jiangyanxi/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/build.cmake:163 (__build_get_idf_git_revision)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:10 (include)

CMake Warning at /home/jiangyanxi/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule
  components/bootloader/subproject/components/micro-ecc/micro-ecc is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/jiangyanxi/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/build.cmake:163 (__build_get_idf_git_revision)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  CMakeLists.txt:10 (include)

CMake Warning at /home/jiangyanxi/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/jiangyanxi/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/build.cmake:163 (__build_get_idf_git_revision)
  /home/jiangyanxi/esp/esp-idf/tools/cmake/idf.cmake:50 (__build_init)
[... 99 lines cut ...]
-- Check for working C compiler: /home/jiangyanxi/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/jiangyanxi/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/jiangyanxi/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/jiangyanxi/esp/esp-idf/examples/bluetooth/esp_ble_mesh/aligenie_demo/sdkconfig
Loading defaults file /home/jiangyanxi/esp/esp-idf/examples/bluetooth/esp_ble_mesh/aligenie_demo/sdkconfig.defaults...
CMake Error at ../../../../tools/cmake/component.cmake:305 (message):
  Include directory
  '/home/jiangyanxi/esp/esp-idf/components/mbedtls/mbedtls/include' is not a
  directory.
Call Stack (most recent call first):
  ../../../../tools/cmake/component.cmake:477 (__component_add_include_dirs)
  ../../../../components/mbedtls/CMakeLists.txt:4 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/home/jiangyanxi/esp/esp-idf/examples/bluetooth/esp_ble_mesh/aligenie_demo/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

cuityanxi, 2021-04-30:
-Subproject commit f31694c9f1746ba189a4bcae2e34db15135ddb22
+Subproject commit 58384fb6af92e7952b1534acb3004924f7ea0565-dirty
diff --git a/components/bootloader/subproject/components/micro-ecc/micro-ecc b/components/bootloader/subproject/components/micro-ecc/micro-ecc
index d037ec895..24c60e243 160000
--- a/components/bootloader/subproject/components/micro-ecc/micro-ecc
+++ b/components/bootloader/subproject/components/micro-ecc/micro-ecc
@@ -1 +1 @@
-Subproject commit d037ec89546fad14b5c4d5456c2e23a71e554966
+Subproject commit 24c60e243580c7868f4334a1ba3123481fe1aa48-dirty
diff --git a/components/bt/host/nimble/nimble b/components/bt/host/nimble/nimble
index 1b808fcb8..f04c92400 160000
--- a/components/bt/host/nimble/nimble
+++ b/components/bt/host/nimble/nimble
@@ -1 +1 @@
-Subproject commit 1b808fcb809c6ceb74c83c0a09f07a63580ca91c
+Subproject commit f04c92400a44ee16e6ee6c02e9247f6d3e213dda-dirty
diff --git a/components/cbor/tinycbor b/components/cbor/tinycbor
--- a/components/cbor/tinycbor
+++ b/components/cbor/tinycbor
@@ -1 +1 @@
-Subproject commit 7c349dbb6b8d76db39383b226d3ebdf59b8ab37d
+Subproject commit 7c349dbb6b8d76db39383b226d3ebdf59b8ab37d-dirty
diff --git a/components/esp_wifi/lib b/components/esp_wifi/lib
index 0c2939b21..e4415c133 160000
--- a/components/esp_wifi/lib
+++ b/components/esp_wifi/lib
@@ -1 +1 @@
-Subproject commit 0c2939b21a816e7f28403a9e69cd257d12817260
+Subproject commit e4415c13367c1dcff2d8e3d7e34052391c71eeff-dirty
diff --git a/components/esptool_py/esptool b/components/esptool_py/esptool

cuityanxi, 2021-04-30:
jiangyanxi@GMPC2020050018:~/esp/esp-idf$ git submodule update
fatal: 仓库 '../../espressif/esp32-bt-lib.git' 不存在
fatal: 无法克隆 '../../espressif/esp32-bt-lib.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/bt/controller/lib_esp32'
克隆 'components/bt/controller/lib_esp32' 失败。按计划重试
fatal: 仓库 '../../espressif/esp32c3-bt-lib.git' 不存在
fatal: 无法克隆 '../../espressif/esp32c3-bt-lib.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/bt/controller/lib_esp32c3_family'
克隆 'components/bt/controller/lib_esp32c3_family' 失败。按计划重试
fatal: 仓库 '../../ThrowTheSwitch/CMock.git' 不存在
fatal: 无法克隆 '../../ThrowTheSwitch/CMock.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/cmock/CMock'
克隆 'components/cmock/CMock' 失败。按计划重试
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/coap/libcoap'...
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/json/cJSON'...
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/lwip/lwip'...

fatal: unable to access 'https://github.com/espressif/esp-lwip.git/': gnutls_handshake() failed: Error in the pull function.
fatal: 无法克隆 'https://github.com/espressif/esp-lwip.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/lwip/lwip'
克隆 'components/lwip/lwip' 失败。按计划重试
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/mbedtls/mbedtls'...

fatal: unable to access 'https://github.com/espressif/mbedtls.git/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
fatal: 无法克隆 'https://github.com/espressif/mbedtls.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/mbedtls/mbedtls'
克隆 'components/mbedtls/mbedtls' 失败。按计划重试
fatal: 仓库 '../../espressif/openthread.git' 不存在
fatal: 无法克隆 '../../espressif/openthread.git' 到子模组路径 '/home/jiangyanxi/esp/esp-idf/components/openthread/openthread'
克隆 'components/openthread/openthread' 失败。按计划重试
正克隆到 '/home/jiangyanxi/esp/esp-idf/components/tinyusb/tinyusb'...

    git子库有问题？

zhanzhaocheng, 2021-05-06:
Since it is very slow to access github in China, you can use the following methods:
1. Use the offline version to install esp-idf: https://dl.espressif.com/dl/esp-idf/?idf=4.4
2. Use gitee: https://gitee.com/EspressifSystems/esp-idf?_from=gitee_search

cuityanxi, 2021-05-07:
gitee解决

closed 2021-05-07 by cuityanxi