espressif/esp-csi #15: [Question] By which API is the rx gain fixed?

JoohyunKang, 2021-05-12:
Hi, thank you for making wonderful CSI project. And I have some questions.

As far as I know, rx gain should be fixed (= Automatic gain control should be disabled) to obtain meaningful CSI data which can detect human movement.
Otherwise, CSI would be almost constant regardless of human movement.

To check whether there exist API which does the function of fixing rx gain, I looked up project codes (e.g., esp-csi/include/esp_radar.h), but there is no API whose name includes "fix" or "rx gain". 

So I'd like to ask 
1. whether rx gain fixing function is implemented (and used) or not, and 
2. which API contains the Rx gain fixing function.

Thanks for reading!

zhanzhaocheng, 2021-05-13:
The RX gain of our RF is fixed, and the TX power can be adjusted by `esp_err_t esp_wifi_set_max_tx_power(int8_t power);`

JoohyunKang, 2021-05-14:
Thanks for the reply.

As far as I know, Wi-Fi station should adjust its rx gain with regard to Wi-Fi AP's RSSI to communicate with the AP stably, and vice versa.

Then, there would be an API that fixes rx gain to specific value with regard to connected Wi-Fi device's RSSI.

If the function exist, I guess the function would be done in esp_wifi_radar_init() or esp_wifi_start(), which are executed after esp_wifi_start().

So what I'd like to ask are:
1. whether the rx gain value is constant value initially(?), or is fixed to specific value?
2. If it is fixed to specific value, then what is the crietria?
3. If it is fixed to specific value, then which API does the function?

Thanks for reading!!!

JoohyunKang, 2021-05-14:
> The RX gain of our RF is fixed, and the TX power can be adjusted by `esp_err_t esp_wifi_set_max_tx_power(int8_t power);`

Thanks for the reply.

As far as I know, Wi-Fi station should adjust its rx gain with regard to Wi-Fi AP's RSSI to communicate with the AP stably, and vice versa.

Then, there would be an API that fixes rx gain to specific value with regard to connected Wi-Fi device's RSSI.

If the function exist, I guess the function would be done in esp_wifi_radar_init() or esp_wifi_start(), which are executed after esp_wifi_start().

So what I'd like to ask are:

whether the rx gain value is constant value initially(?), or is fixed to specific value?
If it is fixed to specific value, then what is the crietria?
If it is fixed to specific value, then which API does the function?
Thanks for reading!!!

zhanzhaocheng, 2021-05-17:
The default RX gain is automatic. PHY layer will automatically gain the access information, so that the device can receive signals of any size. However, if you fix the RX gain, it will cause packet loss. For example, if the RX gain is small, the received RSSI is small, the data will be lost

You can use `wifi_init` After, call `rom_. rx_ gain_ Force '

```c
ESP_ ERROR_ CHECK(esp_ wifi_ init(&cfg));
extern esp_ err_ t rom_ rx_ gain_ force(int force_ en, int force_ value); // force_ value
rom_ rx_ gain_ force(true, 79);
```

UncleBigLu, 2022-01-17:
According to my test, if you want to obtain channel change through CSI amplitude change, you MUST first call rom_rx_gain_force to turn of auto gain control otherwise the CSI amplitude would be almost constant.

strongerux, 2022-02-10:
> 默认 RX 增益是自动的。PHY层会自动获取接入信息，使设备可以接收任意大小的信号。但是，如果您固定 RX 增益，则会导致丢包。比如RX增益小，接收到的RSSI小，数据就会丢失
> 
> 可以使用`wifi_init`后，调用`rom_。rx_gain_force '
> 
> ```c
> ESP_ERROR_CHECK (esp_wifi_init ( &cfg))
>  ; extern esp_err_t rom_rx_gain_force （ int force_en，int force_value）；// force_value 
> rom_rx_gain_force ( true , 79 ) ;
> ```
Can you tell me where is this api(rom_rx_gain_force )?thx

zhanzhaocheng, 2022-02-11:
This API definition is in ROM, it can be used after adding function declaration

strongerux, 2022-02-11:
> This API definition is in ROM, it can be used after adding function declaration
I tried,but:
FAILED: console_test.elf
cmd.exe /C "cd . && C:\Users\123\.espressif\tools\riscv32-esp-elf\esp-2021r1-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc -nostartfiles -march=rv32imc --specs=nosys.specs @CMakeFiles\console_test.elf.rsp -o console_test.elf  && cd ."
c:/users/123/.espressif/tools/riscv32-esp-elf/esp-2021r1-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(app_main.c.obj): in function `wifi_init_sta':
a:\aithinker\workspace\esp-csi-master\examples\console_test\build/../main/app_main.c:127: undefined reference to `rom_rx_gain_force'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

zhanzhaocheng, 2022-02-11:
What chip are you using, currently ESP32/ESP32S2/ESP32S3 are supported
[image]

strongerux, 2022-02-11:
I get it.

zhanzhaocheng, 2023-05-31:
@UncleBigLu  Sorry no one got back to you earlier.

*bool* manu_scale:
For there is a rounding operation to generate the CSI data, to keep the best CSI precision, we'll left shifting all the subcarriers with a same shift value while without making any subcarrier to be saturated.
Setting "manu_scale=ture" is used when you care about the time-variation of CSI's amplitude between frames, but needing to set a proper "shift" parameter, RX gain information also need to be considered. it's not recommended to use if you not quite sure about whether the value can cover your application's scenario.
If you just care about the relative amplitude or phase between subcarriers of certain frame, suggest to use the "manu_scale=false".

closed 2023-06-12 by zhanzhaocheng