espressif/esp-csi #20: Cannot connect to softAP when first connect to a TPLINK Wi-Fi AP

erusyd, 2021-09-22:
Steps to reproduce:
(1) open a terminal 1, type
     sta tplink_ssid tplink_password
(2) open another terminal 2, type
     ap csi_softap
(3) in terminal 1, type
     sta csi_softap
resulted an endless loop.

zhanzhaocheng, 2021-09-22:
The logic you executed is wrong. One device cannot connect to two APs at the same time. Please refer to the following operations:

Steps to reproduce:
(1) open a terminal 1, type
sta tplink_ssid tplink_password
ap csi_softap
(2) open another terminal 2, type
sta csi_softap

erusyd, 2021-09-23:
Thanks for your reply. Is there any way in monitor mode to connect to one AP first, then disconnect from it, after that connect to another AP? What are the operations needed to disconnect from the first AP in order to connect to the second AP successfully?

erusyd, 2021-09-23:
By the way, I have two ESP32 boards, terminal 1 connect to board1, terminal 2 connect to board 2.

erusyd, 2021-09-23:
I use board1 to connect to a TPLINK AP first to sync SNTP time, then I want to keep the synced local time and connect to board2's softAP to do experiment.

zhanzhaocheng, 2021-09-26:
> By the way, I have two ESP32 boards, terminal 1 connects to board1, terminal 2 connects to board2.

You need to call `esp_wifi_stop()` before reconfiguring other Wi-Fi

> I use board1 to connect to a TPLINK AP first to sync SNTP time, then I want to keep the synced local time and connect to board2's softAP to do experiment.

If you want to achieve your board1 synchronization requirements, you need to enable the NATP service on Board2
Or there is Board2 synchronized time, indirectly transmitted to Board1 via Wi-Fi

erusyd, 2021-09-27:
Thanks. I have managed to sync timestamp from Board2 to Board1 through TCP communication.

closed 2021-09-27 by erusyd