espressif/esp-csi #25: Missing cmd_system.h file

dazzalicious, 2021-10-30:
I am new to programming the ESP32-C3 chip and want to get the CSI data for distance ranging between two chips. I've tried to clone your repo for some ideas but I am getting this error trying to flash the console_test program to my chip. I can see a system_cmd.c file in your main folder but no cmd_system.h file.

[image]

dazzalicious, 2021-10-30:
It seems that simply commenting out the #include cmd_system.h lines allows this to work.

zhanzhaocheng, 2021-12-17:
This is caused by a change in the version of ESP-IDF.
He only affects the console:https://github.com/espressif/esp-idf/tree/master/examples/system/console

closed 2023-05-31 by zhanzhaocheng