espressif/esp-csi #27: On timestamp interval between successive CSI feedback

erusyd, 2021-11-10:
I use
_ping 192.168.4.1 -i 0.2_
to set the time interval to 0.2 seconds, but I get the following time intervals by differentiating successive CSI feedback local_timestamp:

[image]

As you can see, most intervals are approximately 0.2s. What I can understand are those ~0.4s and ~0.6s intervals which may be caused by extra ping time waiting or packet loss, but I can't figure out why there are many ~0.01s intervals. Can I simply remove those samples?

kallu1294, 2021-12-10:
> I use _ping 192.168.4.1 -i 0.2_ to set the time interval to 0.2 seconds, but I get the following time intervals by differentiating successive CSI feedback local_timestamp:
>
> [image]
>
> As you can see, [...]

Hi 

For this we need two esp32 board right.? one for AP and another one of STA so we can get the CSI DATA 

thanks in advance

erusyd, 2021-12-13:
@kallu1294 Although you can use another commercial AP, but it is recommended to use two esp32 boards to get stable and consistent CSI data.

closed 2021-12-14 by erusyd

kallu1294, 2021-12-14:
> @kallu1294 Although you can use another commercial AP, but it is recommended to use two esp32 boards to get stable and consistent CSI data.

Thanks for the reply 
Is there any method to download this CSI _DATA..??

kallu1294, 2021-12-16:
> I use _ping 192.168.4.1 -i 0.2_ to set the time interval to 0.2 seconds, but I get the following time intervals by differentiating successive CSI feedback local_timestamp:
>
> [image]
>
> As you can see, [...]

Hello sir
 how can we see the received CSI_DATA like this.??

erusyd, 2021-12-17:
Use ctrl+t, ctrl+l to log the data. You should read the readme of this repository first.