espressif/esp-csi #29: Generating CSI data into graphs

kallu1294, 2021-12-17:
Hi everyone 
Now i have a CSI data so i need to plot those data into graphs how can i do that anybody can suggest me how can i do that..??
And how can differentiate imaginary and real data ..??

zhanzhaocheng, 2021-12-17:
Please refer [here](https://github.com/espressif/esp-csi/tree/master/examples/console_test/tools)

kallu1294, 2021-12-17:
> Please refer [here](https://github.com/espressif/esp-csi/tree/master/examples/console_test/tools)

Thanks for the reply :-)

kallu1294, 2021-12-17:
> Please refer [here](https://github.com/espressif/esp-csi/tree/master/examples/console_test/tools)

Already i have csi_data in csv file which i did from other source(repo) and now i need to plot waveform(analyze) for those data how can i do that any suggestions ...??
And i went through your suggestions but i didn't get anything there,means if i run the commands which are present that link it is shown error as bash:  ./analysis_csi_data.py : no such file or directory

zhanzhaocheng, 2021-12-20:
Each channel frequency response of sub-carrier is recorded by two bytes of signed characters. The first one is imaginary part and the second one is real part.
refer to: https://github.com/espressif/esp-csi/issues/24

closed 2023-05-31 by zhanzhaocheng