espressif/esp-csi #33: Unable to configure CSV file

Greycat3, 2022-01-18:
When I was trying to implement the CSI demo example, I followed the instruction on [this link](https://github.com/espressif/esp-qcloud#3-%E6%9E%84%E5%BB%BA%E7%83%A7%E5%BD%95%E8%BF%90%E8%A1%8C%E5%B7%A5%E7%A8%8B). However, I don't know how to configure the CSV file, I chose partition table in menuconfig and it was shown as followed. I can't find this path ./config/partition_table/partitions_4MB.csv in my workspace. Can anyone tell me how to configure this CSV file?
[image]

zhanzhaocheng, 2022-01-18:
path see: esp-csi/examples/connect_qcloud/components/esp-qcloud/config/partition_table/partitions_4MB.csv

Greycat3, 2022-01-19:
> path see: esp-csi/examples/connect_qcloud/components/esp-qcloud/config/partition_table/partitions_4MB.csv

Thanks for replying. But my question is I have nothing under the path esp-csi/examples/connect_qcloud/components/. Do I need to create it myself or did I miss any step that can create config/partition_table/partitions_4MB.csv under the folder of componenets？

zhanzhaocheng, 2022-01-19:
It should be because your submodule has not been downloaded, please run the following command:
```shell
cd esp-csi
git submodule update --init --recursive
```

closed 2023-05-31 by zhanzhaocheng