espressif/esp-csi #35: Esp32 detect the change of human breathing？

zhy102, 2022-02-25:
Thank you for the information, I would like to ask: **Could the csi extracted from esp32 detect the change of human breathing? Could adding an external antenna improve the sensitivity a lot, and could the sampling rate of csi be set?**

zhanzhaocheng, 2022-02-28:
- Breath testing is theoretically possible
- External antenna can improve sensitivity
- CSI sampling rate can be set

zhy102, 2022-02-28:
您发给我的邮件我已收到。

zhy102, 2022-02-28:
OK, thanks. Best regards!

kavya1229, 2022-04-23:
How can we detect breathing from wifi csi?

zhanzhaocheng, 2022-04-24:
You can view relevant papers: https://ieeexplore.ieee.org/search/searchresult.jsp?newsearch=true&queryText=wifi%20csi%20%20breath

zenbooster, 2023-02-08:
Is it possible multi-person Respiration Sensing, as described here, but with multiple esp32 boards:
https://www.researchgate.net/publication/343361467_MultiSense_Enabling_Multi-person_Respiration_Sensing_with_Commodity_WiFi
?

closed 2023-05-31 by zhanzhaocheng