espressif/esp-csi #36: esp-csi wifiradar inside, can someone add an API to supply the buffer from external malloc?

novflying, 2022-02-27:
Hi everybody, I noticed that there are frequently realloc calls inside the csi library, which will fire a lot of warning messages as the system memory goes fragmented, finally the esp-csi library won't have any chance to alloc the big continneous memory - more than 32K as I can see from the warning log output.
I am wondering is it possible to supply another API that user can pass their own malloced memory which will be used dedicated only for esp-csi library, so it won't fail to alloc memory due to memory fragmentation.

zhanzhaocheng, 2022-02-28:
Thank you very much for your suggestion, due to previous limitations of our algorithm a large memory block was needed to buffer the data. We are currently optimizing it

novflying, 2022-02-28:
Yes I can see from the warning log that it's kept reallocting a buffer of size 32868 bytes, this allocation would fail frequently when other modules in the system are eating out the heap memory. So for now a simple way of walkaround would be allocate this buffer statically and keept it inside the csi library.

zhanzhaocheng, 2023-05-31:
Hello，@novflying 

The latest version has fixed this problem. Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-31 by zhanzhaocheng