espressif/esp-csi #37: Configuring incomplete, errors occurred!

zhy102, 2022-03-08:
The errors is coming when i called "idf.py menuconfig:

```
Executing action: menuconfig
Running cmake in directory c:\espressif\frameworks\esp-idf-v4.3.2\console_test\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 c:\espressif\frameworks\esp-idf-v4.3.2\console_test"...
-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at C:/Espressif/frameworks/esp-idf-v4.3.2/tools/cmake/component.cmake:224 (message):
  CMake Error at C:/Espressif/frameworks/esp-idf-v4.3.2/CMakeLists.txt:2
  (project):

    project command is not scriptable

  Call Stack (most recent call first):

    C:/Espressif/frameworks/esp-idf-v4.3.2/tools/cmake/scripts/component_get_requirements.cmake:97 (include)
    C:/Espressif/frameworks/esp-idf-v4.3.2/tools/cmake/scripts/component_get_requirements.cmake:115 (__component_get_requirements)

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v4.3.2/tools/cmake/build.cmake:420 (__component_get_requirements)
  C:/Espressif/frameworks/esp-idf-v4.3.2/tools/cmake/project.cmake:399 (idf_build_process)
  CMakeLists.txt:23 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v4.3.2/console_test/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

C:\Espressif\frameworks\esp-idf-v4.3.2\console_test>cmake
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>
  cmake [options] -S <path-to-source> -B <path-to-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Run 'cmake --help' for more information.

C:\Espressif\frameworks\esp-idf-v4.3.2\console_test>cmake --version
cmake version 3.16.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

zhanzhaocheng, 2022-03-15:
You can try to clear the temporary files generated by the compilation to see if it can be solved
```shell
idf.py fullclean
rm -rf build
python-clean
```

NJUPT-Sivan, 2022-03-17:
I also encountered this problem with consent, can anyone tell me how to solve this problem? I desperately need your help

zhanzhaocheng, 2022-03-17:
Does [get-started](https://github.com/espressif/esp-csi/tree/master/examples/get-started) compile normally?

closed 2022-04-25 by zhy102

zhy102, 2022-10-11:
您发给我的邮件我已收到。