espressif/esp-csi #40: ping timeout error

zhy102, 2022-03-10:
Esp32-idf:v4.4
Device: esp32-wroom-32u.
Os: win10

I (42554) esp_netif_handlers: sta ip: 192.168.1.102, mask: 255.255.255.0, gw: 192.168.1.1

```
csi> W (52704) ping_cmd: From 192.168.1.1 icmp_seq=14 timeout
W (57094) ping_cmd: From 192.168.1.1 icmp_seq=58 timeout
W (60214) ping_cmd: From 192.168.1.1 icmp_seq=89 timeout
```

zhanzhaocheng, 2022-03-15:
This is due to the poor Wi-Fi network environment, so that ping packets are lost

zhy102, 2022-10-11:
Ok. I have another question about the same frequency jammer. The communication frequency band of ESP32 is around 2.4ghz. When there is a 2.4ghz wifi device or Bluetooth device nearby, can the CSI data collected be stable?

zhanzhaocheng, 2022-10-12:
I have actually tested that the CSI data is stable even in an office environment where there is a lot of wireless signal interference
For better accuracy, the ping interval is therefore 10ms. You can increase the ping interval according to project requirements to reduce packet loss

closed 2023-05-31 by zhanzhaocheng