espressif/esp-csi #41: configuration parameters to the firmware

zhy102, 2022-03-10:
Thank you for the code, I would like to ask: how to write the following configuration parameters to the firmware to burn into the esp32 device.
‘’‘
Steps to test CSI                                |
 |                                                                     |
 |  1. Print 'help' to gain overview of commands                       |
 |     csi> help                                                       |
 |  2. Start SoftAP or Sta on another ESP32/ESP32S2/ESP32C3            |
 |     csi>     sta myssid  mypassword                                      |
 |     csi> ap myssid mypassword                                          |
 |  3. Run ping to test                                                |
 |     csi> ping <192.168.4.2> -c <100> -i <10>                         |
 |  4. Configure CSI parameters                                        |
 |     csi> csi -m <mac(xx:xx:xx:xx:xx:xx)> -l <len>                   |
 |  5. Configure CSI parameters                                        |
 |     csi> detect -a <absolute_threshold> -r <relative_threshold>
‘’’

zhanzhaocheng, 2022-03-15:
This example is for interactive testing, you can develop based on [get-started](https://github.com/espressif/esp-csi/tree/master/examples/get-started) first

zhy102, 2022-03-23:
> This example is for interactive testing, you can develop based on [get-started](https://github.com/espressif/esp-csi/tree/master/examples/get-started) first

Thank you for your advice, but I still want to continue to use the code in "https://github.com/espressif/esp-csi/tree/master/examples/console_test/main". Already found in the firmware app_main.c file where the mac is modified to filter the CSI, but how to configure to write sta commands and ping commands to the firmware? This is very important to me, could you help me again?

zhanzhaocheng, 2022-03-23:
@zhy102 Can you reply to the solution and close it under the solved problem? So that I can quickly understand what problems still exist?

You only need to call `esp_console_run()` in [app_main](https://github.com/espressif/esp-csi/blob/master/examples/console_test/main/app_main.c#L204), just write the command you want to execute, as shown below:
```c
    int ret;
    esp_err_t err = esp_console_run("sta csi", &ret);
```

zhy102, 2022-03-24:
> @zhy102 Can you reply to the solution and close it under the solved problem? So that I can quickly understand what problems still exist?
>
> You only need to call `esp_console_run()` in [...]

OK， thanks.

closed 2022-03-24 by zhy102