espressif/esp-csi #42: pip errors

zhy102, 2022-03-19:
OS is windows10.

```
Collecting PyQt5-Qt5>=5.15.2
  Downloading PyQt5_Qt5-5.15.2-py3-none-win_amd64.whl (50.1 MB)
     ----------------------- ---------------- 30.0/50.1 MB 12.2 kB/s eta 0:27:17
ERROR: Exception:
Traceback (most recent call last):
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\urllib3\response.py", line 438, in _error_catcher
    yield
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\urllib3\response.py", line 519, in read
    data = self._fp.read(amt) if not fp_closed else b""
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\cachecontrol\filewrapper.py", line 90, in read
    data = self.__fp.read(amt)
  File "http\client.py", line 458, in read
  File "http\client.py", line 502, in readinto
  File "socket.py", line 669, in readinto
  File "ssl.py", line 1241, in recv_into
  File "ssl.py", line 1099, in read
socket.timeout: The read operation timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\cli\base_command.py", line 167, in exc_logging_wrapper
    status = run_func(*args)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\cli\req_command.py", line 205, in wrapper
    return func(self, options, args)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\commands\install.py", line 339, in run
    requirement_set = resolver.resolve(
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\resolver.py", line 94, in resolve
    result = self._result = resolver.resolve(
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 481, in resolve
    state = resolution.resolve(requirements, max_rounds=max_rounds)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 373, in resolve
    failure_causes = self._attempt_to_pin_criterion(name)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 213, in _attempt_to_pin_criterion
    criteria = self._get_updated_criteria(candidate)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 204, in _get_updated_criteria
    self._add_to_criteria(criteria, requirement, parent=candidate)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\resolvelib\resolvers.py", line 172, in _add_to_criteria
    if not criterion.candidates:
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\resolvelib\structs.py", line 151, in __bool__
    return bool(self._sequence)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 155, in __bool__
    return any(self)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 143, in <genexpr>
    return (c for c in iterator if id(c) not in self._incompatible_ids)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\found_candidates.py", line 47, in _iter_built
    candidate = func()
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\factory.py", line 215, in _make_candidate_from_link
    self._link_candidate_cache[link] = LinkCandidate(
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 288, in __init__
    super().__init__(
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 158, in __init__
    self.dist = self._prepare()
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 227, in _prepare
    dist = self._prepare_distribution()
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\resolution\resolvelib\candidates.py", line 299, in _prepare_distribution
    return preparer.prepare_linked_requirement(self._ireq, parallel_builds=True)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\operations\prepare.py", line 487, in prepare_linked_requirement
    return self._prepare_linked_requirement(req, parallel_builds)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\operations\prepare.py", line 532, in _prepare_linked_requirement
    local_file = unpack_url(
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\operations\prepare.py", line 214, in unpack_url
    file = get_http_url(
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\operations\prepare.py", line 94, in get_http_url
    from_path, content_type = download(link, temp_dir.path)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\network\download.py", line 146, in __call__
    for chunk in chunks:
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\cli\progress_bars.py", line 304, in _rich_progress_bar
    for chunk in iterable:
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_internal\network\utils.py", line 63, in response_chunks
    for chunk in response.raw.stream(
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\urllib3\response.py", line 576, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\urllib3\response.py", line 541, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "contextlib.py", line 131, in __exit__
  File "c:\espressif\python_env\idf4.4_py3.8_env\lib\site-packages\pip\_vendor\urllib3\response.py", line 443, in _error_catcher
    raise ReadTimeoutError(self._pool, None, "Read timed out.")
pip._vendor.urllib3.exceptions.ReadTimeoutError: HTTPSConnectionPool(host='files.pythonhosted.org', port=443): Read timed out.
```

zhy102, 2022-03-20:
> 
This live image is stuck after some waveforms appear.

zhanzhaocheng, 2022-03-21:
The baud rate of serial port can be set to 2Mbps at most. For the setting method, see: [How to change esp32 console baudrate](https://github.com/espressif/esp-csi/tree/master/examples/console_test#how-to-change-esp32-console-baudrate)

zhanzhaocheng, 2022-03-21:
> > 
> 
> This live image is stuck after some waveforms appear.

Do you scale the graph? Under what circumstances can it be reproduced and what is the probability of reproduction? Will it work after you restart the python script?

closed 2023-05-31 by zhanzhaocheng