espressif/esp-csi #43: is it possible to collect CSI in monitor mode?

gxhen, 2022-03-21:
Hi, I am implementing the example in /get-started. I am writing to ask if it is possible to run the ESP32 in monitor mode rather than STA or AP, which extracts CSI from all packets in the air. 

Thanks in advance.

zhanzhaocheng, 2022-03-22:
You can get all the CSI packets in the air. You just need to comment [`csi_recv.c's equipment filtering](https://github.com/espressif/esp-csi/blob/master/examples/get-started/csi_recv/main/app_main.c#L53-L55) is OK

```c
// examples/get-started/csi_recv/main/app_main.c#L53-L55
//    if (memcmp(info->mac, CONFIG_CSI_SEND_MAC, 6)) {
//        return;
//    }
```

zhanzhaocheng, 2022-03-24:
If the problem has been solved, please close it, thank you