espressif/esp-csi #44: phase waveform instability

zhy102, 2022-03-22:
Hi, I managed to get the CSI matrix of 256 bytes and then I plotted the amplitude and phase of these 128 subcarriers.
 However, the amplitude waveform is working well, but the phase waveform feels strange: even the phase waveform of the CSI acquired by the unmanned environment has a large variation in amplitude. 
Here is my matlab code to extract the phase information: `phase=unwrap(angle(CSI_complex)); `
 Can you tell me the reason for the phase instability?

gxhen, 2022-03-22:
signal phase is prone to be affected by phase noise and the wireless channel. It is reasonable to have unstable phases in the collected CSI.

zhy102, 2022-03-22:
The CSI amplitude signal from the same sample is very good. Is it because the CSI phase of esp32 device contains a lot of random noise like the CSI phase extracted from Intel 5300 network card, so the phase information cannot be used directly? You can get rid of that noise by linear transformation and so on. Right?

zhanzhaocheng, 2023-05-31:
Hello, @zhy102 

Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-31 by zhanzhaocheng

ThETyMaNp, 2024-07-09:
Hello,

I've encountered a similar issue: I anticipated stable CSI amplitude in a static environment for activity detection. However, I've noticed significant fluctuations, including:

A noticeable mutation in the ESP32's received CSI amplitude, which doesn't follow a Gaussian distribution, as seen in Figures 1 and 2.
[image]
[image]
Variations in the signal's overall amplitude over time, even without people present, as shown in Figure 3.
[image]

I'm uncertain if this is due to AGC effects or underestimating environmental changes.

Your assistance is greatly appreciated.
Thank you.

MacChu0315-Espressif, 2024-07-10:
1. Due to the characteristics of electromagnetic waves, the perception range may be larger than you anticipated, and a shielding box would be a more ideal testing environment.
2. Regarding your doubts about AGC, #185 might be helpful to you.