espressif/esp-csi #45: Which subcarriers are invalid when first_word_invalid is true?

gxhen, 2022-03-23:
Hi,

The ESP32 official document indicates if  'first_word_invalid' field of 'wifi_csi_info_t' is true, then the first four bytes of CSI data is invalid due to the hardware limitation of ESP32. 

Q1: The CSI data collected by ESP32 starts from subcarrier 0 - 31, and the first four bytes corresponds to two subcarriers (after merging the imaginary and real parts). Does this mean subcarrier 0 and subcarrier 1 are invalid when 'first_word_invalid' is true? In my case I always have a very high absolute value at subcarrier 0.

Q2:  Will ESP32-S3 and ESP32-C3 have the same issue?

Any help will be appreciated!

zhanzhaocheng, 2022-03-24:
There are 56 subcarriers in each group of CSI

However, there are 52 subcarriers in lltf, of which 4 subcarriers are used for pilot. For effective subcarrier selection, please refer to the following:
```python
csi_data_vaid_index = []
csi_data_vaid_index += [i for i in range(1,27)]      # 26
csi_data_vaid_index += [i for i in range(38, 64)]    # 26
csi_data_vaid_index += [i for i in range(65, 93)]    # 28
csi_data_vaid_index += [i for i in range(100, 128)]  # 28
```

JunyiXb, 2022-07-27:
Hi,I‘am recently researching about csi of ep32 acquisition mechanism,can you recommend some materials for me.I get a table from https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/wifi.html#id57 as follow,I cannot understand  the mean of this table,expecially in the subcarrier index part,which is same among LLTF,HT-LTF,and  STBC-HT-LTF。

[image]

zhanzhaocheng, 2023-05-31:
@JunyiXb Sorry no one got back to you earlier.

Here are some Wi-Fi CSI (Channel State Information) related materials along with their corresponding links for further reference:
1. Academic Papers: https://ieeexplore.ieee.org/search/searchresult.jsp?newsearch=true&queryText=wifi%20csi
2. Development Resources and Tools: https://dhalperi.github.io/linux-80211n-csitool/

closed 2023-06-12 by zhanzhaocheng