espressif/esp-csi #48: How to process CSI data and understand the meaning of the collected 256-byte CSI data

NJUPT-Sivan, 2022-04-07:
If I want to use the collected CSI data to do work related to human activity recognition such as work ,sit, wave..., how do I tag the collected CSI data? What is the meaning of the collected 256 bytes of CSI_DATA? Does it represent 128 subcarriers?

zhanzhaocheng, 2022-04-11:
Yes, channel state information generally refers to all information that can reflect the physical properties of the channel, including multipath delay, Doppler frequency shift, beamforming vector and other multipath effects, which are manifested as amplitude attenuation, phase offset and time delay in the time domain，See: https://ieeexplore.ieee.org/search/searchresult.jsp?newsearch=true&queryText=WIFI%20CSI

KOOKOKOK, 2022-05-24:
你好，上面的连接里有很多篇论文，其中那一篇是是有关介绍csi_data的

closed 2023-05-31 by zhanzhaocheng