espressif/esp-csi #51: Yes, channel state information generally refers to all information that can reflect the physical properties of the channel, including multipath delay, Doppler frequency shift, beamforming vector and other multipath effects, which are manifested as amplitude attenuation, phase offset and time delay in the time domain，See: https://ieeexplore.ieee.org/search/searchresult.jsp?newsearch=true&queryText=WIFI%20CSI

KOOKOKOK, 2022-05-24:
Yes, channel state information generally refers to all information that can reflect the physical properties of the channel, including multipath delay, Doppler frequency shift, beamforming vector and other multipath effects, which are manifested as amplitude attenuation, phase offset and time delay in the time domain，See: https://ieeexplore.ieee.org/search/searchresult.jsp?newsearch=true&queryText=WIFI%20CSI

_Originally posted by @zhanzhaocheng in https://github.com/espressif/esp-csi/issues/48#issuecomment-1094630450_

KOOKOKOK, 2022-05-24:
你好，上面的连接里有很多篇论文，其中那一篇是是有关介绍csi_data的

zhanzhaocheng, 2022-05-25:
It's all about csi_data analysis,
The first one is imaginary part and the second one is real part. See:https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-channel-state-information

closed 2022-06-02 by KOOKOKOK