espressif/esp-csi #52: By default it showing 40 Mhz but in CSI table it is showing 20 Mhz

pampariumeshchandra, 2022-05-31:
I am trying to collect CSI data from 2 ESP32 Wrover B.By default it is showing 40mhz in the settings but when I am collecting the CSI data it is showing 0 in bandwidth(CSW)  (0 for 20 Mhz and 1 for 40 Mhz). Could you please help me in finding the mistake.Below is the attachment of CSI data.

CSI_DATA,12,1a:00:00:00:00:00,rssi--41,rate-11,sig_mode-1,mcs-0,cwb-0,smoothing-1,not_sounding-1,aggregation-0,stbc-0,fec_coding-0,sgi-1,noise_floor--93,ampdu_cnt-0,channel-11,secondary_channel-2,timestamp-131164,ant-0,sig_len-44,rx_state-0,128,1,"[44,-64,2,0,0,0,-4,4,-13,19,-13,20,-13,21,-12,22,-12,23,-12,25,-12,26,-12,27,-12,28,-11,28,-11,30,-11,31,-11,32,-10,32,-11,33,-10,35,-9,36,-8,36,-9,36,-9,37,-8,38,-7,40,-7,40,-6,40,-5,42,-6,43,-6,44,-6,45,-3,23,-3,46,-4,46,-4,47,-4,48,-3,49,-2,50,-2,50,-2,50,-2,50,-2,52,-2,53,-3,53,-4,53,-4,53,-3,54,-4,55,-5,55,-6,56,-6,56,-6,54,-8,53,-9,52,-11,52,-13,52,-14,51,-14,50,-15,49,-16,49,-4,12,0,0,0,0]"
[image]

zhanzhaocheng, 2022-06-01:
You need to checkout the ESP-IDF to the master branch

pampariumeshchandra, 2022-06-02:
HI!!Thanks for the input.Tried it but I am getting error that this script was called from a virtual environment,can not create a virtual environment again.I uninstalled and installed again and kept in master branch I am still getting 20Mhz.

jonathanmuller, 2022-06-02:
Hi, 

Here are some path you can try :

- Verify 40mHz band is really used : Connecting a recent device (computer, phone, ...) and monitor if it uses the 20mHz or 40mHz
- Verify 40mHz is correctly displayed by your code : Go in a busy location and record all CSI frames, does any use the 40mHz