espressif/esp-csi #53: 关于数据的采集

KOOKOKOK, 2022-06-01:
在get-started/csi_recv中，输出的长度为256的数组中哪些是表示子载波振幅哪些是表示子载波的相位

zhanzhaocheng, 2022-06-01:
Each channel frequency response of sub-carrier is recorded by two bytes of signed characters. The first one is imaginary part and the second one is real part.

KOOKOKOK, 2022-06-02:
我查看了get-started案例，发现显示的曲线数据是从esp上获得的复数的模，我在这里想问一下你上面所说的实部和虚部是代指的接受到的载波的信息（比如实部代表振幅，虚部代表角频率），还是说接受到的那个复数就是载波的频率。

zhanzhaocheng, 2022-06-05:
> Note: Please ask questions in English so that others can quickly find questions

CSI is actually the discrete sampling of CFR on different subcarriers.
CSI → related to the amplitude and phase of OFDM subcarriers → the amplitude and phase information of subcarriers is the sampling of CFR spectrum

KOOKOKOK, 2022-06-05:
For this sentence，“CSI is actually the discrete sampling of CFR on different subcarriers.”，I think that CSI is actually the discrete sampling of CIR on different subcarriers instead of CFR.

In other words, I think the data received from the port through the python file（csi_catch.pycsi_catch.py） should be time domain information rather than frequency domain information.

zhanzhaocheng, 2022-06-05:
If you still have doubts, please read this document carefully: [WiFi Sensing with Channel State Information](https://www.cs.wm.edu/~yma/files/WiFiSensing_YongsenMa_authorversion.pdf)

closed 2023-05-31 by zhanzhaocheng