espressif/esp-csi #57: country code of WiFi

zhy102, 2022-06-12:
I uesd the example of "get-started", but i want to know that how to set the country code of WiFi and confirm the current country code？ Please help me. thanks.

zhanzhaocheng, 2022-06-12:
[esp_err_t esp_wifi_set_country_code(const char *country, bool ieee80211d_enabled)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_country_codePKcb)

zhy102, 2022-06-12:
> [esp_err_t esp_wifi_set_country_code(const char *country, bool ieee80211d_enabled)](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv425esp_wifi_set_country_codePKcb)

In the "app_main.c" file of the firmware of the sending end, this sentence of "esp_err_t esp_wifi_set_country_code(const char *country, bool ieee80211d_enabled)" is added, but how to change the value of the country code, such as "JP"? 

```
void app_main()
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    wifi_init();
	//const char *country= ”JP”,;
    esp_err_t esp_wifi_set_country_code(const char *country, bool ieee80211d_enabled);

 
    ESP_ERROR_CHECK(esp_now_init());
    ESP_ERROR_CHECK(esp_now_set_pmk((uint8_t *)"pmk1234567890123"));

    esp_now_peer_info_t peer = {
        .channel   = CONFIG_LESS_INTERFERENCE_CHANNEL,
        .ifidx     = WIFI_IF_STA,    
        .encrypt   = false,   
        .peer_addr = {0xff, 0xff, 0xff, 0xff, 0xff, 0xff},
    };
    ESP_ERROR_CHECK(esp_now_add_peer(&peer));

    ESP_LOGI(TAG, "================ CSI SEND ================");
    ESP_LOGI(TAG, "wifi_channel: %d, send_frequency: %d, mac: " MACSTR,
             CONFIG_LESS_INTERFERENCE_CHANNEL, CONFIG_SEND_FREQUENCY, MAC2STR(CONFIG_CSI_SEND_MAC));

    for (uint8_t count = 0; ;++count) {
        esp_err_t ret = esp_now_send(peer.peer_addr, &count, sizeof(uint8_t));

        if(ret != ESP_OK) {
            ESP_LOGW(TAG, "<%s> ESP-NOW send error", esp_err_to_name(ret));
        }

        vTaskDelay(pdMS_TO_TICKS(1000 / CONFIG_SEND_FREQUENCY));
    }
```

zhanzhaocheng, 2023-05-31:
@zhy102  Sorry no one got back to you earlier. You just need to use the following settings on it
```c
esp_wifi_set_country_code( ”JP”,  false);
```

closed 2023-06-12 by zhanzhaocheng