espressif/esp-csi #58: Is it possible to get sequence numbers along with CSI data?

gxhen, 2022-06-14:
I would like to ask is it possible to return/get the sequence number of the received packet in the CSI callback function 'wifi_csi_rx_cb' along with the CSI data? I cannot find relevant information in the document. Does ESP32 support this officially?

Any help would be highly appreciated.

zhanzhaocheng, 2022-06-15:
Each channel frequency response of sub-carrier is recorded by two bytes of signed characters. The first one is imaginary part and the second one is real part. [Wi-Fi Channel State Information](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=csi#wi-fi-channel-state-information)

gxhen, 2022-06-16:
Sorry I did not state my question clearly. The sequence control field is a 16-bit segment in the 80211 MAC header. As shown in the MATLAB documentation (https://www.mathworks.com/help/wlan/ug/802-11-mac-frame-decoding.html), it is located after address 3. 

In fact, my question is can we obtain the 80211 MAC header in the CSI callback function? Many thanks for your help.

zhanzhaocheng, 2023-05-31:
@gxhen Sorry no one got back to you earlier. 

This feature is currently not supported, we will support it in esp-idf release/v5.1

gxhen, 2023-06-01:
@zhanzhaocheng Many thanks for your reply. I have checked the released v5.1/v5.1-beta1 document (https://docs.espressif.com/projects/esp-idf/en/v5.1-beta1/esp32/api-reference/network/esp_wifi.html#_CPPv415wifi_csi_info_t) and there is still no sequence number available in the 'wifi_csi_info_t' struct. 

Will this be released in the updated documentation? Thanks for answering.

zhanzhaocheng, 2023-06-02:
@gxhen Currently this feature is still in internal testing, if it is updated to github I will notify you immediately

closed 2023-06-12 by zhanzhaocheng