espressif/esp-csi #60: wifi_radar_cb no callback succeeded

homent-hutu, 2022-06-15:
step:
1、 wifi_config -s TPLinkHome -p 1234567890.
2、 ping -i 10 -c 0

device error log：
W (81912) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81925) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81931) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81932) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81940) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81946) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81956) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81964) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81971) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81980) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (81987) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82005) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82015) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82027) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82036) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82045) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82055) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82061) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1
W (82068) esp_radar: value fail, len: 376, secondary_channel: 1, sig_mode: 1, cwb: 0, stbc: 1

zhanzhaocheng, 2022-06-16:
This is caused by the wrong judgment when filtering the CSI data. I have fixed it. You can change the channel of the router or use the ESP32 as the AP for testing. I will update it on github as soon as possible.

bilalfarooq002, 2022-06-29:
Hi, @zhanzhaocheng. I am also facing a similar problem, I have tried to set the router channel same as secondry_channel shown in the device logs of @homent-hutu, but it is still giving the same error.

step:
1、 wifi_config -s <ssid_of_router> -p <password>.
2、 ping

my device logs:
`(126513) ping_cmd: From 192.168.10.1 icmp_seq=36 timeout
W (126516) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0
W (126520) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0
W (126526) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0
W (126531) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0
W (126536) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0
W (126541) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0
CSI_DATA,570,0,0,24:d3:f2:95:dd:d5,-56,11,1,7,0,1,1,1,0,0,0,-95,1,1,1,126294475,0,145,0,216,0,"[-19,9,-20,7,-20,6,-20,4,-21,3,-21,3,-21,3,-21,3,-21,2,-21,2,-22,2,-22,2,-22,2,-22,1,-22,1,-22,1,-22,1,-22,2,-22,2,-22,2,-21,3,-21,3,-21,4,-20,4,-20,5,-19,5,-19,7,-19,8,-19,8,-19,9,-18,10,-18,10,-18,11,-18,11,-17,12,-17,13,-17,13,-17,13,-17,13,-16,13,-16,13,-16,13,-16,13,-15,13,-15,13,-14,13,-14,13,-14,13,-15,12,-15,12,-16,11,-17,10,-13,13,-16,11,-18,9,-19,7,-19,6,-19,5,-20,4,-20,3,-20,3,-20,2,-20,2,-20,2,-21,1,-21,1,-22,1,-22,1,-22,1,-22,1,-22,2,-21,2,-21,2,-21,2,-20,3,-20,3,-20,4,-20,4,-19,4,-19,5,-18,7,-18,7,-18,8,-18,9,-18,9,-17,10,-17,10,-17,11,-17,11,-16,12,-16,12,-16,12,-16,12,-16,12,-15,13,-15,13,-15,12,-14,12,-14,12,-14,12,-14,12,-14,12,-14,12,-15,11,-16,11,-17,10,-17,7,-18,5]"
CSI_DATA,571,0,0,24:d3:f2:95:dd:d5,-56,11,1,7,0,1,1,1,0,0,0,-95,1,1,1,126297495,0,106,0,216,0,"[-21,-6,-21,-8,-20,-9,-19,-11,-19,-12,-18,-12,-18,-13,-18,-13,-18,-13,-18,-13,-18,-14,-19,-14,-19,-14,-19,-15,-19,-15,-19,-15,-19,-14,-19,-14,-19,-14,-19,-14,-19,-13,-19,-12,-19,-12,-19,-11,-19,-10,-19,-10,-20,-8,-21,-8,-21,-7,-21,-6,-22,-6,-22,-5,-22,-4,-22,-4,-23,-3,-23,-2,-23,-2,-23,-2,-23,-2,-22,-2,-22,-2,-22,-1,-22,-1,-21,-1,-21,-1,-21,-1,-20,-1,-20,-1,-21,-1,-21,-2,-21,-3,-20,-5,-20,1,-21,-2,-21,-5,-20,-7,-19,-9,-19,-10,-18,-11,-17,-11,-17,-12,-17,-12,-18,-12,-18,-13,-18,-13,-18,-13,-18,-14,-18,-14,-18,-14,-18,-14,-18,-14,-18,-14,-18,-13,-18,-13,-19,-12,-18,-11,-18,-11,-18,-10,-18,-10,-19,-9,-20,-7,-20,-7,-20,-6,-20,-6,-21,-5,-21,-5,-21,-4,-22,-3,-22,-3,-22,-2,-22,-2,-22,-2,-22,-2,-22,-1,-21,-1,-21,-1,-21,-1,-21,-1,-20,-1,-20,0,-20,0,-20,-1,-20,-1,-20,-2,-20,-3,-20,-5,-19,-6,-18,-8]"
CSI_DATA,572,0,0,24:d3:f2:95:dd:d5,-56,11,1,7,0,1,1,1,0,0,0,-95,1,1,1,126446672,0,94,0,216,0,"[17,-13,17,-12,18,-10,19,-9,19,-8,19,-7,19,-7,19,-7,20,-6,20,-6,21,-6,21,-6,21,-6,21,-6,21,-6,21,-6,21,-6,21,-6,21,-7,20,-7,20,-7,20,-8,19,-8,18,-8,18,-9,17,-9,17,-11,17,-12,16,-12,16,-13,15,-13,15,-14,15,-14,14,-15,14,-16,13,-16,13,-16,13,-16,13,-16,13,-16,12,-16,12,-16,12,-16,11,-16,11,-16,10,-16,10,-15,10,-15,11,-15,12,-15,13,-14,13,-13,11,-16,13,-15,16,-13,17,-11,17,-10,18,-8,18,-8,18,-7,19,-7,19,-7,19,-7,19,-6,20,-6,20,-6,20,-6,20,-6,20,-5,20,-5,20,-6,20,-6,20,-6,19,-7,19,-7,18,-8,18,-8,18,-8,17,-8,17,-9,16,-11,16,-11,15,-12,15,-12,15,-13,14,-14,14,-14,14,-14,13,-15,13,-15,13,-15,13,-15,13,-15,12,-15,12,-16,11,-16,11,-16,11,-16,10,-15,10,-15,10,-15,10,-15,10,-15,11,-14,12,-14,13,-13,15,-10,15,-8]"
CSI_DATA,573,0,0,24:d3:f2:95:dd:d5,-56,11,1,7,0,1,1,1,0,0,0,-95,2,1,1,126448691,0,94,0,216,0,"[20,-7,21,-5,21,-3,21,-1,21,0,20,0,21,1,21,1,21,1,21,1,22,1,22,1,22,1,22,2,22,2,22,2,22,2,22,2,22,2,22,1,21,1,21,0,20,-1,20,-1,20,-2,19,-3,20,-4,20,-5,20,-5,20,-6,20,-7,19,-7,19,-8,19,-9,19,-9,18,-10,18,-10,18,-10,18,-11,18,-11,17,-11,17,-10,17,-11,16,-11,16,-11,16,-10,16,-10,15,-10,16,-10,16,-10,17,-9,17,-8,17,-11,19,-9,20,-6,20,-4,20,-2,20,0,20,0,19,1,20,2,20,2,20,2,20,2,21,2,21,2,21,2,21,2,21,3,21,3,21,3,21,3,21,2,21,2,20,1,20,1,19,0,19,-1,19,-1,19,-1,19,-3,19,-4,19,-5,19,-5,19,-6,19,-7,19,-7,18,-8,18,-8,18,-9,17,-9,18,-9,17,-10,17,-10,17,-10,17,-10,16,-10,16,-10,16,-10,15,-10,15,-10,15,-9,15,-9,16,-9,16,-8,17,-7,17,-3,17,-2]"
W (127368) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0
W (127545) ping_cmd: From 192.168.10.1 icmp_seq=43 timeout
W (127548) esp_radar: value fail, len: 384, secondary_channel: 1, sig_mode: 1, cwb: 1, stbc: 0`

in between i get this CSI_DATA, i also tried to use the esp_csi_tool.py but i got any data about corr. 
screenshot is given below:

[image]

zhanzhaocheng, 2022-07-12:
@bilalfarooq002 This issue has been fixed, please verify it after update, And the current detection algorithm is optimized

bilalfarooq002, 2022-08-29:
Hi again @zhanzhaocheng, I was actually able to solve the problem in the previous version of CSI tool. but in newer version I am facing a similar issues which I am not been able to resolve.

I have complied the console test example and I am trying to test the new algorithm but there seems to be bug in code maybe that I am able to see the raw data on csi-tool but there is no data related to movement or human status.

Here are the steps that I have followed to get there,

1.  First I upload the console-test code and started the csi-tool.
2.  There is bug in the csi-tool that it sends the `radar --csi_output_type LLFT --csi_output_format base64` command too early I think. Anyways I have to send this command manually using csi-tool in order to change the format of data. if I don't send this command, data is sent in wrong format and is never displayed in the csi-tool.
3.  After that I start receiving the raw data, and then I start the calibration. once that is done nothing happens in the **radar model windows** as shown in the image below
[image]

4.  You might be seeing the ping timeout but the ping is working fine, as only the timeouts are displayed in the terminal and not the successful ping attempts.

I have tested and found out that `wifi_radar_cb()` callback is nevet called. As I don't have the radar library code to troubleshoot the problem maybe you can help me if I am missing something?

zhanzhaocheng, 2023-05-31:
@bilalfarooq002 Sorry no one got back to you earlier.

There is no problem with your test process. There was a problem of ping failure in the previous version of esp-idf. Please update esp-idf and esp-csi to test again

closed 2023-06-12 by zhanzhaocheng