espressif/esp-csi #61: CSI resolution and phase unwrapping

NaivelyWritten, 2022-06-16:
Hi,

I have 2 questions for the project.

1. In my experiment all CSI values seems to fall in the range of -16-16, is this intended? A byte should have a range of -128-127?
2. According to the literature the CSI in a ideal channel should have a linear phase if unwrapped. However in the measurement I got it is almost a fixed phase with some variation. Is this because you already did linear phase removal in the closed source blob?

Thanks a lot!

zhanzhaocheng, 2022-07-12:
The value of csi is between -128 and 127, and the received signal amplitude and phase are related

NaivelyWritten, 2022-07-12:
Thanks for the reply! However the problem I have is that it is indeed -128-127 for HT-LTF, but for L-LTF it's only -16-16.
And also the order is different when associated/unassociated.

我的测试里面 L-LTF 部分只有 -16-16 的范围，另外如果和AP处于连接状态的时候 CSI 的顺序是反的。

zhanzhaocheng, 2022-07-13:
You can observe the value of L-LTF by changing the distance between the device and the AP.
You can try the [console_test](https://github.com/espressif/esp-csi/tree/master/examples/console_test) example to observe the subcarrier changes

closed 2023-05-31 by zhanzhaocheng