espressif/esp-csi #64: CSI_Data 绘图

tanghy2016, 2022-07-13:
在下面的代码中, ```csi_vaid_subcarrier_index```的第0个元素为0, 也就是说构建的第一个复数```complex```其实是```csi_raw_data```的第0个数和最后一个数. 这里是否正确呢？

```python

csi_vaid_subcarrier_index = []
csi_vaid_subcarrier_color = []
color_step = 255 // (28 // CSI_VAID_SUBCARRIER_INTERVAL + 1)

# LLTF: 52
csi_vaid_subcarrier_index += [i for i in range(0, 26, CSI_VAID_SUBCARRIER_INTERVAL)]     # 26  red
csi_vaid_subcarrier_color += [(i * color_step, 0, 0) for i in range(1,  26 // CSI_VAID_SUBCARRIER_INTERVAL + 2)]
csi_vaid_subcarrier_index += [i for i in range(26, 52, CSI_VAID_SUBCARRIER_INTERVAL)]    # 26  green
csi_vaid_subcarrier_color += [(0, i * color_step, 0) for i in range(1,  26 // CSI_VAID_SUBCARRIER_INTERVAL + 2)]

...

def csi_data_handle(self, data):
    csi_raw_data = json.loads(data['data'])
    
    # Rotate data
    g_csi_phase_array[:-1]     = g_csi_phase_array[1:]
    g_rssi_array[:-1]          = g_rssi_array[1:]
    g_radio_header_pd.iloc[1:] = g_radio_header_pd.iloc[:-1]

    for i in range(CSI_DATA_COLUMNS):
        data_complex = complex(csi_raw_data[csi_vaid_subcarrier_index[i] * 2],
                               csi_raw_data[csi_vaid_subcarrier_index[i] * 2 - 1])
        g_csi_phase_array[-1][i] = np.abs(data_complex)

    g_rssi_array[-1]         = data['rssi']
    g_radio_header_pd.loc[0] = data[1:]

    return
```

zhanzhaocheng, 2022-07-14:
@tanghy2016 should be the zeroth number and the first number,  Each channel frequency response of sub-carrier is recorded by two bytes of signed characters. The first one is imaginary part and the second one is real part

tanghy2016, 2022-07-14:
好的，谢谢回复，那这应该是个BUG，需要修复，就是源码中的```esp_csi_tool.py```里面

closed 2022-07-14 by tanghy2016