espressif/esp-csi #65: Using two esp32

Tisha36, 2022-07-20:
I want to use two esp32 for CSI collection. I I will use one esp as transmitter and another one as receiver. So when i go through the interface window ,in the router connection window ,what should i give in the field of  router ssid and password part.
Thanks.

zhanzhaocheng, 2022-07-21:
No need to enter router information if using two ESP32s
Make sure your project is up to date

Gowtham-Gorajana, 2022-10-15:
@zhanzhaocheng Can you please provide more information with detailed explanation on how we can use two esp32s with the given source files so that we can detect any human passing between the transmitter and the receiver.
Any paper or a video link would be very much helpful.
Thanks

zhanzhaocheng, 2023-05-31:
@Gowtham-Gorajana Sorry no one got back to you earlier.

You can run `examples/get-started/csi_recv` on one device and `examples/get-started/csi_send` on the other, and it will meet your needs

closed 2023-06-12 by zhanzhaocheng