espressif/esp-csi #66: CSI Data

tiyuuu, 2022-07-23:
From CSI data we get bandwidth, noise, aggregation, csi value ,mac etc. But how these value can help for plant monitoring or human activity recognition or in other projects.
I am new to this field. So I can't understand.
Thanks in advance.

zhanzhaocheng, 2022-07-24:
Through further processing of this information, it can help us improve the accuracy of detection
For example, through the Mac address, we can obtain the source device of the CSI data, and process the CSI information sent by different devices separately; through the noise, we can obtain the interference of the network environment, and judge whether the network environment meets the detection requirements when the device is arranged. noise for corresponding data processing

tiyuuu, 2022-07-24:
> Through further processing of this information, it can help us improve the accuracy of detection For example, through the Mac address, we can obtain the source device of the CSI data, and process [...]

And from amplitudes and phase for each frame of CSI data, what can we understand?

zhanzhaocheng, 2022-07-24:
Channel state information can reflect the physical properties of the channel, including multipath delay, Doppler frequency shift, beamforming vector and other multipath effects in the time domain as amplitude attenuation, phase offset and time delay

tiyuuu, 2022-07-30:
Thanks. Can  you suggest how to preprocess the CSI data to remove environmental noise?
I found this type of CSV file of CSI data, where I get noise floor
[image]

zhanzhaocheng, 2022-08-02:
You can process CSI data through Kalman, IIR and other filtering algorithms. At the same time, it is strongly recommended that you use the development board of esp32-s3 external antenna for development. Esp32-s3 Wi Fi has better RF performance and better directivity of external antenna

closed 2023-05-31 by zhanzhaocheng