espressif/esp-csi #69: CSI数据集相关疑问

tanghy2016, 2022-08-22:
[image]

你好，关于CSI数据，我遇到一下几个问题：
1. 上图中"辅助信道, 空, 上, 下"分别表示什么意思？
2. L-LTF, HT-LTF, STBC-HT-LTF, 区别是什么？
3. 为什么我使用ESP32获取到的CSI数据"总字节数", 与上图不一样呢，用不同的路由器获取到的字节数不一样, 有 552, 384, 216，这时我如何通过索引得到 L-LTF, HT-LTF, STBC-HT-LTF 子载波呢？
4. 在源码的python脚本(examples/console_test/esp_csi_tool.py)中说使用的子载波有什么特别的含义吗？为什么间隔步长5取一个子载波呢？

```python
# Remove invalid subcarriers
# secondary channel : below, HT, 40 MHz, non STBC, v, HT-LFT: 0~63, -64~-1, 384

CSI_VAID_SUBCARRIER_INTERVAL = 5

csi_vaid_subcarrier_index += [i for i in range(
    0, 26, CSI_VAID_SUBCARRIER_INTERVAL)]

csi_vaid_subcarrier_index += [i for i in range(
    26, 52, CSI_VAID_SUBCARRIER_INTERVAL)]
```

zhanzhaocheng, 2022-08-22:
1.  second channel: for HT20, second is ignored, for HT40, second is the second channel
2. See: [High Throughput Long Training Field (HT-LTF)](http://rfmw.em.keysight.com/wireless/helpfiles/n7617a/n7617a.htm#high_throughput_long_training_field.htm)
3. The number of sub-carriers is different for different communication methods, see: Wi-Fi Channel State Information
4. Displaying multiple subcarriers at the same time will freeze in the Python GUI, so every 5th subcarrier is displayed

dfx1822375, 2022-10-09:
> [image]
> 
> 你好，关于CSI数据，我遇到一下几个问题：
> 
> 1. 上图中"辅助信道, 空, 上, 下"分别表示什么意思？
> 2. L-LTF, HT-LTF, STBC-HT-LTF, 区别是什么？
> 3. 为什么我使用ESP32获取到的CSI数据"总字节数", 与上图不一样呢，用不同的路由器获取到的字节数不一样, 有 552, 384, 216，这时我如何通过索引得到 L-LTF, HT-LTF, STBC-HT-LTF 子载波呢？
> 4. 在源码的python脚本(examples/console_test/tools/esp_csi_tool.py)中说使用的子载波有什么特别的含义吗？为什么间隔步长5取一个子载波呢？
> 
> ```python
> # Remove invalid subcarriers
> # secondary channel : below, HT, 40 MHz, non STBC, v, HT-LFT: 0~63, -64~-1, 384
> 
> CSI_VAID_SUBCARRIER_INTERVAL = 5
> 
> csi_vaid_subcarrier_index += [i for i in range(
>     0, 26, CSI_VAID_SUBCARRIER_INTERVAL)]
> 
> csi_vaid_subcarrier_index += [i for i in range(
>     26, 52, CSI_VAID_SUBCARRIER_INTERVAL)]
> ```

您好， @tanghy2016 ,我针对CSI做了一些研究，可以留个邮箱交流下相关问题么

closed 2023-05-31 by zhanzhaocheng