espressif/esp-csi #70: examples/console_test is compile error

michan06, 2022-08-22:
Thank you for sharing great project!!

I want to run console_test on esp32-c3, but set-target  esp32c3 and build, I encountered following error.
> ninja: error: '****/esp-csi/lib/esp32c3/libesp-csi.a', needed by 'console_test.elf', missing and no known rule to make it

After this commit  
https://github.com/espressif/esp-csi/commit/d183220e97c23b6ded528bf8ae2cc0c0d56ab9db ,
esp32-c3 and esp32-s2's "libesp-csi.a" was deleted.

[image]

Please tell me that console_test supports esp32-c3 or not.

Thank you.

zhanzhaocheng, 2022-08-22:
The processing power of esp32-C3 is very weak, so CSI development based on ESP32-C3 is not recommended, and console_test is not configured for ESP32-C3

michan06, 2022-08-22:
I understand that use of esp-csi on 
esp-c3 in  is not recommended.  Thank you.

zhanzhaocheng, 2023-05-31:
Hello, @michan06 

The latest version has fixed this problem. Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-31 by zhanzhaocheng