espressif/esp-csi #75: problems about ESP connected to commodity WiFi routers

jiechen0809, 2022-09-30:
Hi, when I connected the ESP32s2 with commodity WiFi routers, there are always different types of CSI packets(such as 384 bytes or 612 bytes), which maybe because of STBC or non-STBC. But even though I filter the CSI packets according to the length of CSI, there are always different types of CSI packets frequently because of the long GI or short GI, which will result in the fluctuation of CSI data( no matter amplitude or phase).  This phenomenon will result in the wrong person detection results. And filtering the packets according to length of CSI will lead to the result of no satisfied CSI packets. 
So, is thers any method that can control the type of CSI packets? For example, there are only non-STBC and long GI CSI packets, which can produce stable CSI data.
Another question, is it possible that the product which can get 1*2 tx-rx WiFi link CSI data will  be available soon? 
Thanks. Any help would be highly appreciated.
[image]

jiechen0809, 2022-09-30:
> Hi, when I connected the ESP32s2 with commodity WiFi routers, there are always different types of CSI packets(such as 384 bytes or 612 bytes), which maybe because of STBC or non-STBC. But even [...]

oh，when the ap is also ESP32s2, the phenomenon above-mentioned will decrease a lot of course. I just wonder that if there is any method for the problem in which ESP served as sta and commodity WiFi as AP.

dfx1822375, 2022-10-07:
> 

Hi, can we leave an email to exchange CSI related questions ?

zhanzhaocheng, 2022-10-08:
@jiechen0809 If you want to be compatible with all router models, you can just take the LLTF field of csi, that is, the first 128 bytes of data

[image]

jiechen0809, 2022-10-08:
> @jiechen0809 If you want to be compatible with all router models, you can just take the LLTF field of csi, that is, the first 128 bytes of data
> 
> [image]

Thanks for your reply. What I mean is that even if I just take the LLTF field of csi, the types of csi packets still change frequently. For example, change from non-stbc to stbc or from long GI to short GI，which will lead to the large fluctuation of CSI data(include amplitude and phase). And this phenomenon cannot be solved by filtering simply, because the changes will  last for tens of seconds.

zhanzhaocheng, 2022-10-08:
@jiechen0809 From the theoretical and practical test point of view, there is no problem you mentioned, and some customers also use it like this

LLTF is unaffected with or without HT-LTF, STBC-HT-LTF：
[image]

jiechen0809, 2022-10-09:
> @jiechen0809 From the theoretical and practical test point of view, there is no problem you mentioned, and some customers also use it like this
> 
> LLTF is unaffected with or without HT-LTF, STBC-HT-LTF： [image]

Thanks. I will have a try.

jiechen0809, 2022-10-09:
> Another question, is it possible that the product which can get 1*2 tx-rx WiFi link CSI data will be available soon?
> Thanks. Any help would be highly appreciated.

How about the second question? Is it possible that the product which can get 1*2 tx-rx WiFi link CSI data will be available soon( 2 receive antennas)?
Thanks.

dfx1822375, 2022-10-09:
> @zhanzhaocheng
> 
> How about the second question? Is it possible that the product which can get 1*2 tx-rx WiFi link CSI data will be available soon( 2 receive antennas)? Thanks.
dear @jiechen0809 ：
Hello, I have published some research on CSI signals, and now I want to collect the signals independently, 
can we communicate with each other about the problems?

jiechen0809, 2022-10-09:
> @zhanzhaocheng
> How about the second question? Is it possible that the product which can get 1*2 tx-rx WiFi link CSI data will be available soon( 2 receive antennas)? Thanks.
> dear @jiechen0809 ： [...]

yes, sure.

dfx1822375, 2022-10-09:
> yes, sure.

Thanks，can you give me your email or other contact information?

jiechen0809, 2022-10-09:
> Thanks，can you give me your email or other contact information?

894901552@qq.com

zhanzhaocheng, 2022-10-09:
> @zhanzhaocheng
> 
> How about the second question? Is it possible that the product which can get 1*2 tx-rx WiFi link CSI data will be available soon( 2 receive antennas)? Thanks.

The chip supporting 1*2 Tt-rx is not currently planned, but next year will mass-produce [ESP32-C5](https://www.espressif.com/en/news/ESP32-C5) which supports Wi-Fi 6 and 5.8G, can get more sub-carriers and better performance

zhanzhaocheng, 2022-10-09:
> Thanks，can you give me your email or other contact information?

If you have a specific commercial project in progress, you can submit your needs [here](https://www.espressif.com/en/contact-us/technical-inquiries), and there will be specialized technical staff to connect with you

jiechen0809, 2022-10-09:
> If you have a specific commercial project in progress, you can submit your needs [here](https://www.espressif.com/en/contact-us/technical-inquiries), and there will be specialized technical staff to connect with you

Thanks for your reply. If people want to have a wide and deployable application based on ESP series, there should acquire the  correct phase information of CSI.

closed 2023-05-31 by zhanzhaocheng