espressif/esp-csi #77: Detect any Human Presence Using Two Esp32s

Gowtham-Gorajana, 2022-10-15:
Can you please provide a detailed explanation on how to use the csi data to detect the human presence between the transmitter and the receiver. How we should send the data and set up the equpment and all.
research papers and video links would be helpful
Thanks

zhanzhaocheng, 2022-10-18:
You can try this [example](https://github.com/espressif/esp-csi/tree/master/examples/console_test#esp-csi-console_test)

Gowtham-Gorajana, 2022-10-18:
I am just new to this field and after extracting the CSI data how can we detect the presence of the humans graphically like it should give a spike when ever it detects a human ,etc...

Gowtham-Gorajana, 2022-10-25:
what do we have to do to establish a connection between the two esp32s 
how can we transfer the data between the two

zhanzhaocheng, 2023-05-31:
Hello, @Gowtham-Gorajana 

Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-31 by zhanzhaocheng

Asafe33, 2024-06-13:
@zhanzhaocheng this link is unreachable