espressif/esp-csi #78: Could not off AGC with rom_rx_gain_force function

asleepjoey, 2022-10-20:
I tried to fix rx gain with the solution in the issue #15 
But there is no difference with default. More specifically, csi amplitude is not reduced even if rssi gets lower.
I wonder if I change something wrong.

Board: esp32-s3

esp-idf: v4.4.1

CONFIG_LESS_INTERFERENCE_CHANNEL: 9

Code change:
1. app_main.c in csi_recv.
```
extern esp_err_t rom_rx_gain_force(int force_en, int force_value);
rom_rx_gain_force(true, 40); // I tried the second parameter 20, 40, 60, 79  
```
2. app_main.c in csi_send
```
ESP_ERROR_CHECK(esp_wifi_set_max_tx_power(40));
```

Monitor log:
I (24) boot: ESP-IDF v4.4.1 2nd stage bootloader
I (25) boot: compile time 14:22:54
I (25) boot: chip revision: 0
I (25) boot.esp32s3: Boot SPI Speed : 80MHz
I (25) boot.esp32s3: SPI Mode : DIO
I (26) boot.esp32s3: SPI Flash Size : 2MB
I (27) boot: Enabling RNG early entropy source...
I (27) boot: Partition Table:
I (28) boot: ## Label Usage Type ST Offset Length
I (29) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (29) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (30) boot: 2 factory factory app 00 00 00010000 00100000
I (31) boot: End of partition table
I (32) esp_image: segment 0: paddr=00010020 vaddr=3c070020 size=130a8h ( 77992) map
I (47) esp_image: segment 1: paddr=000230d0 vaddr=3fc94800 size=0414ch ( 16716) load
I (51) esp_image: segment 2: paddr=00027224 vaddr=40374000 size=08df4h ( 36340) load
I (59) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=682cch (426700) map
I (136) esp_image: segment 4: paddr=000982f4 vaddr=4037cdf4 size=07a00h ( 31232) load
I (143) esp_image: segment 5: paddr=0009fcfc vaddr=50000000 size=00010h ( 16) load
I (150) boot: Loaded app from partition at offset 0x10000
I (150) boot: Disabling RNG early entropy source...
I (152) cpu_start: Pro cpu up.
I (152) cpu_start: Starting app cpu, entry point is 0x403751d8
0x403751d8: call_start_cpu1 at /home/asleep/esp/esp-idf/components/esp_system/port/cpu_start.c:160
I (0) cpu_start: App cpu up.
I (161) cpu_start: Pro cpu start user code
I (161) cpu_start: cpu freq: 160000000
I (161) cpu_start: Application information:
I (161) cpu_start: Project name:     csi_recv
I (162) cpu_start: App version:      ff41f7b-dirty
I (162) cpu_start: Compile time:     Oct 19 2022 14:22:50
I (163) cpu_start: ELF file SHA256:  70c5ea4b570630c1...
I (164) cpu_start: ESP-IDF:          v4.4.1
I (164) heap_init: Initializing. RAM available for dynamic allocation:
I (165) heap_init: At 3FC9C3C8 len 00043C38 (271 KiB): D/IRAM
I (166) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (167) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (168) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (169) spi_flash: detected chip: generic
I (170) spi_flash: flash io: dio
W (170) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (172) sleep: Configure to isolate all GPIO pins in sleep state
I (172) sleep: Enable automatic switching of GPIO sleep configuration
I (173) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (209) pp: pp rom version: e7ae62f
I (209) net80211: net80211 rom version: e7ae62f
I (219) wifi:wifi driver task: 3fce5544, prio:23, stack:6656, core=0
I (219) system_api: Base MAC address is not set
I (219) system_api: read default base MAC address from EFUSE
I (219) wifi:wifi firmware version: 63017e0
I (219) wifi:wifi certification version: v7.0
I (219) wifi:config NVS flash: enabled
I (219) wifi:config nano formating: disabled
I (219) wifi:Init data frame dynamic rx buffer num: 128
I (219) wifi:Init management frame dynamic rx buffer num: 128
I (219) wifi:Init management short buffer num: 32
I (219) wifi:Init dynamic tx buffer num: 32
I (219) wifi:Init static tx FG buffer num: 2
I (219) wifi:Init static rx buffer size: 2212
I (219) wifi:Init static rx buffer num: 10
I (219) wifi:Init dynamic rx buffer num: 128
I (219) wifi_init: rx ba win: 6
I (219) wifi_init: tcpip mbox: 32
I (219) wifi_init: udp mbox: 6
I (229) wifi_init: tcp mbox: 6
I (229) wifi_init: tcp tx win: 5744
I (229) wifi_init: tcp rx win: 5744
I (229) wifi_init: tcp mss: 1440
I (229) wifi_init: WiFi IRAM OP enabled
I (229) wifi_init: WiFi RX IRAM OP enabled
I (229) phy_init: phy_version 501,94697a2,Apr  7 2022,20:49:08
I (329) wifi:mode : sta (f4:12:fa:d8:07:24)
I (329) wifi:enable tsf
I (329) wifi:Set ps type: 0
I (339) wifi:enable tsf
I (339) wifi:ic_enable_sniffer

zhanzhaocheng, 2022-10-21:
Currently, ESP32-S3 currently does not support setting AGC on the public version.
I also don't recommend you to fix the AGC, it will bring a lot of other Wi-Fi data sending and receiving problems
Are you currently having any problems with the default automatic AGC? I can give some advice according to your needs

asleepjoey, 2022-10-24:
I want to see the change in CSI(especially amplitude) according to human activity while changing the distance between sender and receiver. It is difficult to test it as CSI amplitude is automatically corrected even if I adjust the sender-receiver distance when AGC is on.

zhanzhaocheng, 2022-10-26:
Did you try to use this [example](https://github.com/espressif/esp-csi/tree/master/examples/console_test) to test? It can display the change of CSI amplitude in real time through GUI interface

closed 2023-05-31 by zhanzhaocheng