espressif/esp-csi #79: abort() was called at PC 0x4015d702 on core 0

gc-night, 2022-10-21:
Hello, I connect router to test console_test example. After i click train_start, abort() was called at PC 0x4015d702 on core 0    
Backtrace:0x40081d02:0x3fff45a00x40088e05:0x3fff45c0 0x4008f922:0x3fff45e0 0x4015d702:0x3fff4650 0x401435c7:0x3fff4670 0x401431ce:0x3fff4690 0x400da264:0x3fff46b0 0x400e332d:0x3fff46d0 0x400e355b:0x3fff4730 0x400d9b3f:0x3fff48d0 
ELF file SHA256: 6f4781e2dcc2e1ce

    then i use gdb for check the problem:
    (gdb) list *0x400d9b3f
0x400d9b3f is in csi_data_handle_task (/Users/zhanzhaocheng/project/esp-csi/esp-csi_release/esp-csi/examples/console_test/components/esp-csi/src/esp_radar.c:413).
408     /Users/zhanzhaocheng/project/esp-csi/esp-csi_release/esp-csi/examples/console_test/components/esp-csi/src/esp_radar.c: No such file or directory.

zhanzhaocheng, 2022-10-21:
Thank you very much for finding this problem, this problem has now been fixed, and it will be updated to GitHub synchronously in the past two days

zhanzhaocheng, 2022-10-26:
@gc-night This bug has been fixed and updated to github, please verify it after the update, if there are still problems, you are very welcome to ask

closed 2023-05-31 by zhanzhaocheng