espressif/esp-csi #81: Consultation on obtaining and storing CSI of multiple ESP32 anchor nodes at the same time

Sunlu2333, 2022-11-09:
I'm trying to reproduce the experimental configuration highlighted in the following paper.
[image]

Refer to the following [project](https://github.com/StevenMHernandez/ESP32-CSI-Tool), I have implemented the CSI acquisition of one sending and one receiving through the active_ Ap and active_ STA, as shown in the figure. 
[image]

However, when applying the passive subproject, I cannot obtain any information through the monitor.
[image]

What I want to do is to implement an AP as the TX and multiple STAs as the RXs(anchor nodes）, and then one ESP32 listens to the CSIs in these links. How can I achieve this and store the CSI data as a csv file? What I have are multiple ESP32-WROOOM-32D boards and.

Thank you very much for your attention and help!

zhanzhaocheng, 2022-11-10:
You can refer to [this document ](https://github.com/espressif/esp-csi/tree/master/examples/get-started#get-started-examples)to try

Sunlu2333, 2022-11-10:
Thank you for your suggestions! But I have a new problem: When I use one ESP32 board as AP and two ESP32 boards as STAs, which ping the same AP at the same time, and the fourth ESP32 board is assigned to sniff mode, I can only receive CSI from AP, and the number of packets received is not stable (All four boards work in channel 4). In detail, I can occasionally receive CSI from two STA, but after a period of time, I can only receive CSI from AP. So does it mean ESP32 will randomly listen to packets from AP or STA when working in sniff mode? What if I only want to listen to CSI from STA?
[image]

zhanzhaocheng, 2022-11-10:
You can use the mac address for filtering, If the received CSI data is not what you want, just discard it

closed 2022-11-18 by Sunlu2333