espressif/esp-csi #88: csi_send ssid and password

vivekrtx, 2022-12-22:
Is there an option to provide ssid and password for the csi_send project? Or, what is the ssid and password set as in csi_send?

zhanzhaocheng, 2022-12-26:
The example of csi_send uses esp-now to send packets, so it does not need to be connected to a router, you just need to let the example of csi_send and the receiving end be on the same channel

vivekrtx, 2023-01-04:
How to selectively connect with one of the two esp32 where both are sending packets? Both are flashed with csi_send and are located at different places. I need to connect to one of them with another esp32 (3rd one) maybe flashed with console_test instead of csi_receive. So I need to assign both of them ssid's to know which one I connected to. How do I assign ssid and password to selectively connect to one of them?

zhanzhaocheng, 2023-01-04:
csi_send uses a different mac address, and the recv end can be filtered by mac address：https://github.com/espressif/esp-csi/blob/master/examples/get-started/csi_send/main/app_main.c#L45

closed 2023-05-31 by zhanzhaocheng