espressif/esp-csi #89: When CSI Demo Example(Wi-Fi radar)  will be available ?

mtibaas, 2022-12-29:
Hello,
when the mobile app will be available ? It's quite interesting and we need to deploy a demo version with our smart plug based on ESP32
https://github.com/espressif/esp-csi/tree/master/examples/connect_qcloud

Thanks in davance for your answer

zhanzhaocheng, 2022-12-29:
qcloud will no longer be maintained, we are currently developing an example based on our own cloud platform (esp-rainmaker)

mtibaas, 2022-12-29:
Ok thanks a lot, it's only available on your cloud platform. I mean - is it possible to integrate it with another cloud platform? I want to setup a demo version in mobile app.
Regards,
Sabri

zhanzhaocheng, 2022-12-30:
It can be connected to any cloud platform, and many of our customers have already connected to their own cloud platform

zhanzhaocheng, 2023-05-31:
Hi @mtibaas 

The latest version has replaced qcloud with RainMaker, and it can work normally on ESP32-C3/ESP32-S3/ESP32-S/ESP32. Closing this issue, feel free to reopen with more updates. Thanks.

closed 2023-05-31 by zhanzhaocheng