espressif/esp-csi #91: Can't get the `console_test` GUI to work

idofr, 2023-01-08:
I flashed my ESP32 WROOM with the firmware in `get-started/csi_rev_router` and it all seems to be working perfectly fine.
Then, I followed the instructions in https://github.com/espressif/esp-csi/tree/master/examples/console_test (my case - only a single esp32). The flashing process finished without any errors, but the GUI does not show anything.
The `log` window is empty and the router doesn't show the device connected.

Since I'm not using the S3, I set the baud flag to 115200, but I also tried the example's 921600. Both didn't work.

Any tips to what I might be doing wrong?

Cheers,

zhanzhaocheng, 2023-01-11:
After you use esp_csi_tool.py to run the GUI, you need to manually press EN on the ESP32 development board. This is because there is a bug in the ESP32 serial port. If you open the serial port through python without restarting the device, garbled characters will appear.

zenbooster, 2023-02-08:
> you need to manually press EN on the ESP32 development board

Doesn't work for me...
(esp32-wroom-32)

zenbooster, 2023-02-08:
If you have CP210x, max baud rate will be 921600, not 2000000!
In examples/console_test/tools/esp_csi_tool.py, function serial_handle:
set baudrate to 921600.

In examples/console_test/sdkconfig.defaults:
change all baud rates from 2000000 to 921600.

Also, you may have to do in examples/console_test/main/app_main.c:
```
#define LED 2

//static led_strip_t *g_strip_handle    = NULL;

esp_err_t led_init()
{
    // Configure pin
    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = (1ULL << LED);
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 0;
    gpio_config(&io_conf);

    gpio_set_level(LED, 1);
    vTaskDelay(500 / portTICK_RATE_MS);
    gpio_set_level(LED, 0);
    return ESP_OK;
}

esp_err_t led_set(uint8_t red, uint8_t green, uint8_t blue)
{
    gpio_set_level(LED, 1);
    vTaskDelay(100 / portTICK_RATE_MS);
    gpio_set_level(LED, 0);

    return ESP_OK;
}

```
After the above procedures, everything began to work for me!

zhanzhaocheng, 2023-02-10:
@zenbooster Thank you very much for finding the root cause of this issue, I will fix it in the next version

xiangshang-kok, 2023-04-05:
@zenbooster sir,I set the baud rate as you instructed.in examples/console_test/main/app_main.c:  should I add your code  here, or delete the original and write it directly as this?

idofr, 2023-04-10:
Sorry for the long absence.
@zenbooster thanks for the explanations, I can confirm that it's working now.
(I replaced the original c code with your code).
Ok to close. 
Cheers,

closed 2023-05-31 by zhanzhaocheng