espressif/esp-csi #92: Which Equipment?

jsl303, 2023-01-09:
I'm hoping someone to help with a few basic questions.
1. Is there esp32-s3 development board that you can simply connect via USB to Raspberry PI without Soldering pins?
2. Is there any particular router brand/model we need to use, or any router would work?
3. Can the router be shared with other clients like phones, or the router should be dedicated for esp32-s3 development board to connect?
Thanks!

zenbooster, 2023-02-08:
Will it work with esp32-wroom-32 devkit?
With TTGO T18 v3.0 or ESP32-DevKitC-DA?

zhanzhaocheng, 2023-02-10:
@jsl303 
1. ESP32-S3-USB-OTG: https://docs.espressif.com/projects/espressif-esp-dev-kits/en/latest/esp32s3/esp32-s3-usb-otg/user_guide.html
2.  any router
3. The router can be shared with other clients such as mobile phones

zhanzhaocheng, 2023-02-10:
> 

All ESP32/ESP32-S3/ESP32-C3 development boards will work

GuoJingtao-1997, 2023-02-24:
> 

Hi. Thanks for sharing this great work! I wonder if this work will support the ESP32-C5 which has dual-band Wi-Fi 6 radio in the future? Thanks for your reply.

GuoJingtao-1997, 2023-02-24:
> > 
> 
> All ESP32/ESP32-S3/ESP32-C3 development boards will work

Hi. Thanks for sharing this great work! I wonder if this work will support the ESP32-C5 which has dual-band Wi-Fi 6 radio in the future? Thanks for your reply.

zhanzhaocheng, 2023-02-27:
@GuoJingtao-1997 Will support, the CSI effect of Wi-Fi 6 and 5.8G will be better

andreac, 2023-03-03:
Any suggestion for a ESP32-S3 with external antenna connection? i'm not able to find on Aliexpress. Any link accepeted

closed 2023-05-31 by zhanzhaocheng