espressif/esp-csi #93: Beacon signal CSI

zikunliu, 2023-01-12:
Is it possible to recevie beacon signal from AP and get CSI of beacon signal? How to implement it? Thank you in advance.

kszlgithub, 2023-03-30:
It is not possible to recevie beacon signal from AP and get CSI of beacon signal.
For beacon frame is a subtype of magenement frames and they transmit at the 802.11b rate. While only transmitions of data frame that have rate of 802.11 g / n standards support OFDM and so that you can get csi data.
By the way, In project esp-csi we use esp-now which bases on action frame (a subtype of Management frame) and it supports transmition rate of 802.11 g / n.

zikunliu, 2023-04-18:
This data was collected in promiscuous mode with channel 2 and fillter MGMT packet.

CSI_DATA,21,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,2130190,0,220,0,128,1, [-36 -64 13 0 8 -25 7 -24 6 -26 3 -28 4 -28 5 -30 7 -30 9 -29 9 -25 10 -23 7 -20 5 -21 4 -18 3 -18 -1 -20 0 -21 -2 -20 -2 -19 -2 -18 -2 -19 -1 -20 -2 -21 0 -21 2 -22 5 -19 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 2 -9 1 -11 1 -12 2 -17 0 -17 3 -18 5 -18 7 -18 8 -20 6 -22 6 -23 8 -22 8 -25 11 -27 11 -23 13 -25 15 -24 11 
-22 13 -21 10 -25 13 -26 13 -25 12 -23 13 -26 11 -27 9 -29]
CSI_DATA,22,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,2232609,0,220,0,128,1, [-36 -64 13 0 26 7 26 4 28 3 27 1 31 1 31 2 33 6 31 5 29 5 25 6 17 2 20 1 20 2 19 0 16 -4 21 -4 22 -7 21 -5 23 
-4 17 -5 19 -4 20 -5 23 -4 21 -1 21 3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 11 1 12 1 14 0 16 1 18 0 18 2 20 4 18 7 22 6 21 6 23 7 26 7 25 10 25 13 24 10 28 11 25 12 25 11 25 12 26 12 24 11 26 11 24 10 27 10 29 9 30 9]
CSI_DATA,23,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,2334996,0,220,0,128,1, [-36 -64 13 0 23 12 23 11 23 8 24 10 26 11 26 8 26 13 25 13 22 16 22 14 20 10 20 10 18 6 18 5 17 5 18 4 20 4 18 4 18 3 16 2 17 3 20 4 18 5 19 7 15 10 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 10 2 12 1 13 0 16 3 17 2 17 3 18 6 19 8 20 8 21 8 22 9 24 9 23 11 24 13 25 14 23 14 22 17 22 13 24 15 23 15 21 16 23 14 22 15 23 17 25 15 28 15]
CSI_DATA,24,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,2461354,0,220,0,128,1, [-36 -64 13 0 12 22 13 21 13 20 14 20 17 20 17 23 16 21 15 22 13 21 12 20 9 15 10 15 11 14 11 12 14 11 14 10 13 10 13 10 14 10 14 9 13 9 15 8 16 11 13 13 11 14 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 5 8 6 8 8 9 9 11 9 12 9 15 8 15 9 18 8 19 7 18 10 21 10 21 9 23 9 24 6 22 8 27 7 25 8 24 5 22 7 25 8 24 8 25 7 24 10 24 
11 24 10 25]
CSI_DATA,25,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,2540159,0,220,0,128,1, [-36 -64 13 0 -7 -27 -8 -26 -9 -28 -11 -26 -9 -27 -12 -29 -8 -30 -8 -30 -5 -28 -6 -26 -6 -21 -5 -19 -7 -20 -9 -18 -11 -17 -12 -18 -11 -16 -11 -17 -10 -16 -11 -15 -11 -17 -13 -16 -11 -17 -10 -19 -7 -20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -7 -10 -7 -10 -9 -12 -9 -15 -10 -17 -9 -18 -10 -20 -7 -21 -6 -22 -6 -24 -7 -25 
-7 -25 -9 -26 -7 -28 -4 -27 -5 -32 -4 -29 -4 -27 -5 -29 -3 -30 -2 -30 -3 -29 -2 -27 -4 -28 -6 -31 -7 -30]
CSI_DATA,26,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,2751909,0,220,0,128,1, [-36 -64 13 0 -15 20 -15 21 -13 24 -11 24 -13 26 -13 26 -13 25 -17 24 -17 22 -16 20 -11 16 -11 16 -9 16 -7 17 -6 19 -7 19 -5 18 -5 17 -5 17 -3 17 -4 17 -5 18 -5 18 -9 19 -10 15 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -5 10 -3 11 -4 13 -5 15 -7 16 -8 16 -9 18 -10 18 -13 19 -13 18 -14 20 -17 21 -17 21 -18 19 -17 20 -22 21 -17 21 -19 21 -18 19 -21 19 -20 18 -20 21 -19 20 -19 22 -18 25 -19 22]
CSI_DATA,27,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,2847008,0,220,0,128,1, [-36 -64 13 0 5 25 5 23 9 24 10 24 12 24 10 27 10 26 8 28 8 25 5 25 5 21 6 17 7 18 7 15 9 16 10 16 11 15 10 14 
11 12 11 14 11 14 11 13 12 16 11 18 6 16 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 5 10 4 10 5 11 8 14 7 15 7 16 6 19 7 19 6 21 5 21 5 23 3 23 5 26 3 26 3 28 3 28 2 27 2 27 1 25 2 29 0 27 2 26 2 25 2 27 5 31 6 29]
CSI_DATA,28,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-70,10,0,0,0,0,0,0,0,-95,0,2,0,2949411,0,220,0,128,1, [-36 -64 13 0 22 1 23 1 24 -2 24 -3 25 -5 28 -6 28 -3 28 -1 25 0 24 2 20 0 18 -3 17 -3 17 -4 16 -6 17 -6 17 -7 
17 -7 15 -7 14 -8 14 -9 16 -8 17 -8 20 -6 18 -3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 10 -2 9 -3 12 -2 15 -4 18 -4 17 0 21 0 19 0 20 4 21 3 22 4 23 5 24 6 24 4 24 5 25 7 24 6 25 4 26 6 26 8 24 8 25 4 24 5 27 4 28 4 29 5]
CSI_DATA,29,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,3051814,0,220,0,128,1, [-36 -64 13 0 -25 -11 -24 -7 -27 -8 -26 -7 -28 -8 -29 -9 -28 -9 -30 -10 -28 -12 -24 -11 -20 -8 -20 -6 -19 -5 -19 -4 -19 -2 -20 -2 -19 1 -19 0 -17 0 -18 0 -19 2 -20 2 -21 -1 -20 -3 -19 -5 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -12 -4 -12 -1 -14 -3 -17 -4 -17 -4 -18 -7 -18 -9 -18 -11 -21 -10 -21 -11 -22 -12 -22 -14 -23 
-12 -24 -13 -22 -15 -24 -17 -22 -17 -22 -14 -21 -16 -23 -17 -22 -16 -22 -14 -23 -15 -24 -16 -26 -16 -24 -16]
CSI_DATA,30,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,3154216,0,220,0,128,1, [-36 -64 13 0 3 -26 5 -23 1 -28 0 -28 2 -30 2 -32 4 -30 6 -31 6 -27 6 -26 3 -23 3 -21 2 -19 -1 -20 -1 -20 -1 -20 -2 -20 -2 -20 -2 -19 -4 -19 -3 -19 -2 -19 -2 -21 0 -21 3 -19 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -2 -10 -3 -11 -2 -15 -2 -16 0 -20 -1 -19 2 -19 1 -20 4 -22 5 -22 3 -22 5 -25 4 -27 6 -27 10 -26 9 -28 6 -25 7 -25 9 -26 9 -27 9 -25 9 -28 9 -25 8 -27 8 -30 6 -29]
CSI_DATA,31,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,3256619,0,220,0,128,1, [-36 -64 13 0 23 10 22 9 26 6 26 6 28 7 29 8 27 12 27 11 23 12 24 12 19 8 18 5 17 5 18 4 19 2 19 2 18 1 17 1 17 1 17 0 17 0 18 -2 19 2 18 4 18 7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 10 2 10 0 12 1 16 2 17 4 17 6 18 8 17 7 17 10 18 9 20 10 21 9 21 11 22 13 20 14 23 16 20 14 21 13 22 14 22 15 21 15 23 15 21 15 22 15 25 16 23 14]
CSI_DATA,32,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,3359022,0,220,0,128,1, [-36 -64 13 0 -13 -23 -14 -24 -18 -23 -18 -23 -18 -24 -20 -26 -17 -27 -15 -28 -14 -23 -11 -25 -10 -20 -11 -17 -12 -15 -13 -16 -16 -16 -15 -15 -15  [... cut ...]
CSI_DATA,42,1c:b9:c4:b5:a9:58,ff:ff:ff:ff:ff:ff,-69,10,0,0,0,0,0,0,0,-95,0,2,0,4588358,0,220,0,128,1, [-36 -64 13 0 -24 9 -23 10 -23 13 -24 13 -24 15 -25 16 -27 14 -27 11 -27 10 -25 8 -19 8 -19 9 -16 10 -15 10 -14 13 -15 13 -14 13 -13 12 -14 12 -12 14 -14 14 -14 15 -16 15 -18 12 -20 10 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 -10 4 -11 5 -12 5 -14 10 -15 9 -18 6 -19 5 -20 5 -20 6 -20 6 -23 6 -24 7 -27 6 -26 5 -25 5 -28 
5 -26 2 -26 4 -24 6 -26 6 -27 3 -26 4 -25 5 -28 7 -28 7 -29 6]

Is this beacon CSI? If is not beacon CSI, what is type of this packet? Thanks.

zhanzhaocheng, 2023-05-31:
@zikunliu Sorry no one got back to you earlier.

They are not beacon CSI, they are packet CSI.  The Beacon sent by the AP is a management packet, which is sent at 11g rate and cannot trigger CSI

closed 2023-06-12 by zhanzhaocheng