espressif/esp-csi #94: understanding the CSI data

olliec28, 2023-02-10:
Can anyone tell me what one row of the CSI data represents is it 1 row = 1 packet of data or does one row = 1 second for example. I am just trying to understand the data in order to feed it into a relevant ML algorithm. If anyone could tell me what algorithms they used as Im struggling to identify suitable ML algorithms at the moment. Thanks

zhanzhaocheng, 2023-02-13:
A line of CSI is generated for each packet of data received
There is no limit to the algorithm of machine learning, such as RNN, CNN, KNN, etc.

olliec28, 2023-02-13:
> A line of CSI is generated for each packet of data received
> There is no limit to the algorithm of machine learning, such as RNN, CNN, KNN, etc.

Will there need to be any signal preprocessing beforehand or simply input the CSI data straight into a neural network?

zhanzhaocheng, 2023-02-13:
You can preprocess the data through algorithms such as Kalman filtering and low-pass filtering to improve the accuracy of recognition

olliec28, 2023-03-09:
do you know which part of the code implements the local timestamp?

zhanzhaocheng, 2023-05-31:
@olliec28 Sorry no one got back to you earlier.

Each frame of CSI information returned by ESP32 contains a timestamp, you can find it in `wifi_csi_info_t>rx_ctrl>timestamp`

closed 2023-06-12 by zhanzhaocheng