espressif/esp-csi #95: real and imaginary part

olliec28, 2023-02-18:
I was just reading that the CSI data comes in 2 bytes as real and imaginary. Can anyone give more detail on how the imaginary part is calculated and should the imaginary part be removed during data pre-processing so that only the real CSI data is being used?

zhanzhaocheng, 2023-02-20:
Each channel frequency response of sub-carrier is recorded by two bytes of signed characters. The first one is imaginary part and the second one is real part.
refer to: https://github.com/espressif/esp-csi/issues/24

olliec28, 2023-02-20:
I know that, that’s what I’ve just stated. But how is the imaginary part calculated?

zhanzhaocheng, 2023-02-21:
I don't understand what you want to ask, can you describe it more clearly?

olliec28, 2023-02-22:
I’m asking where has this imaginary part come from, is it an estimation or guess? Like how is it calculated?

zhanzhaocheng, 2023-02-23:
This is generated at the physical layer. You can refer to [this paper](https://www.cs.wm.edu/~yma/files/WiFiSensing_YongsenMa_authorversion.pdf) for the specific generation process.

closed 2023-05-31 by zhanzhaocheng